kmackay/micro-ecc #17: Problem compiling AVR code for Arduino UNO

dwhire, 2015-01-29:
I have been able to run the C version of micro-ecc on my Arduino. However I get errors when 
compiling the ASM procs. Please help. I don't know AVR ASM. I just would like to use the faster option. Thank you.
Dave Henderson

Compiler errors reported:

C:/Arduino/myArduinoLibs/Ecc/asm_avr.inc:137: error: can't find a register in class 'POINTER_Y_REGS' while reloading 'asm'
C:/Arduino/myArduinoLibs/Ecc/asm_avr.inc:137: error: can't find a register in class 'POINTER_Y_REGS' while reloading 'asm'
(repeats above line 6 times)
C:/Arduino/myArduinoLibs/Ecc/asm_avr.inc:137: error: 'asm' operand has impossible constraints
(repeats above line 6 times)

Code:

static uint8_t vli_add(uint8_t *p_result, uint8_t *p_left, uint8_t *p_right)
{
    uint8_t i = uECC_BYTES;
    uint8_t l_carry = 0;
    uint8_t l_left;
    uint8_t l_right;
 __asm__ volatile (
        "clc \n\t"
        
        "1: \n\t"
        "ld %[left], x+ \n\t"  /* Load left byte. */
        "ld %[right], y+ \n\t" /* Load right byte. */
        "adc %[left], %[right] \n\t" /* Add. */
        "st z+, %[left] \n\t"  /* Store the result. */
        "dec %[i] \n\t"
        "brne 1b \n\t"
        
        "adc %[carry], %[carry] \n\t"    /* Store carry bit in l_carry. */
        
        "sbiw r28, " STR(uECC_BYTES) " \n\t" /* Restore Y */

        : "+z" (p_result), "+x" (p_left), [i] "+r" (i),
            [carry] "+r" (l_carry), [left] "=&r" (l_left), [right] "=&r" (l_right)
        : "y" (p_right)
        : "cc", "memory"
    );

    return l_carry;
}

kmackay, 2015-01-29:
When using the inline assembly for AVR, you need to build with optimizations enabled (-O1 or higher)

closed 2015-02-03 by kmackay

atlazar, 2017-01-06:
I also have Arduino UNO. Constantly get same linker errors (try with -O1 or -Os flags). After many attempts to resolve this, I found by chance what change from

```
#define uECC_SUPPORT_COMPRESSED_POINT 1
```
to
```
#define uECC_SUPPORT_COMPRESSED_POINT 0
```
in uECC.h resolve this issue. Maybe this helps somebody.