kmackay/micro-ecc #19: uECC_verify on hash with only 4 bytes

hetneo, 2015-02-17:
Hi Kenneth,

I need to verify on Cortex M0 the signature of only 4 bytes.

I have the data to verify the signature : 0x11223344 (4 bytes)
I have also the signature 20 bytes (use curve uECC_secp160r1)

Is it possible to use your lib to make this verification ?

If not, do you have an easy solution to do this ?

Thanks in advance

Best regards

kmackay, 2015-02-17:
I'm not sure exactly what you mean. If you mean that the data/hash you are signing is only 4 bytes, then you can just pad out the hash with zeros to make it 20 bytes long. You need to do this the same way on the verifying side.

Bear in mind that if your hash is only 4 bytes long, the probability of collisions is high, and it will be fairly easy for an attacker to create a fake message that produces the same hash (and therefore the signature for the fake message would be "verified").

kmackay, 2015-02-17:
Also, the signature size should be 40 bytes when using secp160r1, so you might want to check your code.

hetneo, 2015-02-18:
Thanks for your quick answer.

Ok for the signature size on 40bytes with secp160r1.

I use SCV Cryptomanager software to try the signature verification with your Micro-ECC code.

I sign data (20 bytes)
11 22 33 44 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
I have this result on 40 bytes :
27 C8 7D 3E EB 9E E8 33 39 ED 51 B0 87 C2 DA F4 D9 A4 8A F1 84 CC DE 7D CB 19 32 0A 9F EF AB 09 53 C7 4F FB E4 98 B2 03
Public Key : 7C 64 75 F5 F8 0E 01 5C A9 42 5A 91 13 D6 AC 3D 36 9C 8B A8 77 BF C8 98 D1 91 16 33 B6 29 BB 14 C4 D5 61 71 CA 1D DC A5

I use uECC_verify on Cortex M3, and all is OK

For my application, I only have 4 bytes of data (11 22 33 44)  and the resulting signature (ECDSA) of 40 bytes and the public key of 40 bytes.

When I use uECC_verify with l_hash = 11 22 33 44 or 11 22 33 44 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 the result is wrong.

I cannot pad the data with 00, because I haven't the private key (the signing is calculated by a third party) and I just have to check the signature.

Best regards

kmackay, 2015-02-18:
OK. I am assuming that SCV CryptoManager is hashing the data (probably using SHA256 or similar) and then using that (truncated to the appropriate size) for signing. You need to hash the message in exactly the same way when doing the verification, and pass the resulting hash to uECC_verify(). I don't know specifically how SCV CryptoManager works, so I can't tell you exactly how to do it.

hetneo, 2015-02-19:
Hi Kenneth,
Forget about SCV CryptoManager ...

To summarize, I have to verify ECDSA signature of 4 data bytes, using secp128r1 (I know it's a bad use of ECDSA, but I'm not the issuer, and I can't change it). I have pubKey, data and signature (s,r), and curve parameters of secp128r1.

The pb is that your uECC_verify() needs a size of uECC_BYTES as input data  "const uint8_t p_hash[uECC_BYTES]". I cannot pad data with 00 because data is signed with exactly 4 bytes long.

So the question is, there is a way to use you uECC_verify with data of 4 bytes long instead of uECC_BYTES long ?

Thank you in advance

Best Regards

kmackay, 2015-02-19:
The data that you are verifying can be of arbitrary length.

When you sign some arbitrary-length data using ECDSA, you first hash the data using a cryptographic hash. Then you take the first n bytes of the hash, where n is the ECC key length. The ECDSA signing algorithm performs various elliptic curve operations on those n bytes (along with the private key), and produces the signature. In micro-ecc those elliptic curve operations are performed by uECC_sign() - you pass in the private key and the n bytes of the hash, and it produces the signature.

When you want to verify an ECDSA signature, you first hash your data using the exact same hashing algorithm that was used to sign it. Then you take the first n bytes of the hash, where n is the ECC key length. The ECDSA verification algorithm performs various elliptic curve operations on those n bytes along with the public key and signature, and determines whether or not the signature is valid. In micro-ecc those elliptic curve operations are performed by uECC_verify() - you pass in the public key, the n bytes of the hash, and the signature, and it returns true or false.

micro-ecc does not specify the hashing algorithm used (since there are many options). It requires you to do the hashing in your own code, and pass the resulting hash in to the uECC_sign() and uECC_verify() functions. You need to use the same hashing algorithm for both signing and verification. If you are trying to verify a signature from somebody else, you need to determine which hashing algorithm they are using and use that.

kmackay, 2015-02-19:
Also, micro-ecc no longer supports secp128r1. The secp128r1 curve parameters make it difficult to implement in the same way that the other curves are implemented. Also, secp128r1 is no longer regarded as secure.

If you want to use secp128r1, you could try using the code in the "old" branch. There is documentation in the ecc.h file. You will still need to hash you data correctly in order to verify the signature.

closed 2015-02-19 by kmackay

hetneo, 2015-02-19:
Sorry I made a mistake, I want to use secp160r1 ! I know secp128r1 is no longer secure. 

Also, I agree with you about hashing data before passing to uECC_verify(). But the pb is that the third party I've the signature data (s and r) from, doesn't use hash alogrithm (bad thing I know), and directly use data (4 bytes long) as input value. 
They have computed uECC_sign  over secp160r1 curve with data (4bytes) as hash value !
And I have to verify this signature !

Is there a way to use your lib with arbitrary length data as input value (as replacement of fixed size hash value) ?

Best regards

kmackay, 2015-02-19:
If the third-party signer is using the 4 byte value directly as a hash
value, you should be able to do the equivalent thing with micro-ecc. The
problem is that you need to know how they use the 4 bytes. In ECDSA, the
hash value is interpreted as a large integer which is used when
calculating/verifying the signature. Typically the hash value is
interpreted as a big-endian integer. If that is what your third party is
doing, it might make sense to prepend your 4 bytes of data with zeros (so
you would create a 20 byte array where the first 16 bytes are 0, and the
last 4 bytes are your data values) and pass that in as the hash value.
However I can't tell you if this will be correct or not; I can't really
help you further without knowing more about the third party that created
the signature.

hetneo, 2015-02-23:
Hi Kenneth,

All works well with your method !
Thank you for your help, it's perfect.

Bye