kmackay/micro-ecc #20: Builds only on a Mac?

jbdatko, 2015-02-18:
It looks like this library builds only on a Mac? For example, on a 64 bit Ubuntu machine I'm trying `emk arch=rpi` and I get the following errors. Maybe I'm not building it correctly?

```
emk (info): Using 12 threads
emk (info): Could not import 'config' from ['/home/jbd/.emk']
emk (info): Entering directory /home/jbd/repos/micro-ecc
emk (info): Entering directory /home/jbd/repos/micro-ecc/test

Build error: Could not call '/Volumes/xtools/arm-none-linux-gnueabi/bin/arm-none-linux-gnueabi-gcc -Wp,-MMD,/home/jbd/repos/micro-ecc/__build__/rpi/dbg/uECC.o.dep -I/home/jbd/repos/micro-ecc -DTARGET_ARCH_RPI=1 -fvisibility=hidden -Wall -Wextra -Wshadow -Werror -Wno-missing-field-initializers -Wno-unused-parameter -Wno-comment -Wno-unused -Wno-unknown-pragmas -g -fomit-frame-pointer -std=c99 -o /home/jbd/repos/micro-ecc/__build__/rpi/dbg/uECC.o -c /home/jbd/repos/micro-ecc/uECC.c' as a subprocess: [Errno 2] No such file or directory
```

kmackay, 2015-02-18:
The emk stuff is set up for my dev environment. I recommend that you just copy (or symlink) uECC.h, uECC.c, and optionally the asm.inc for your platform into your project and build them using your own build system. The reason is that micro-ecc is designed for embedded systems, so typically it will be built using a cross-compiler with options specifically for the target board.

Alternatively you could just edit emk_project.py and change the paths for the compiler and linker to point to your Raspberry Pi cross-compiler (in the setup_linux_rpi() function).

jbdatko, 2015-02-18:
Makes sense. Thanks for the response.

closed 2015-02-18 by jbdatko