kmackay/micro-ecc #22: uECC_compress/uECC_decompress with ECC_CURVE set to uECC_secp256r1

totalslacker, 2015-03-01:
I'm working with the uECC_secp256r1 curve and am seeing a case where a public key that is compressed then decompressed no longer seems to be correct. The actual byte array is different and any attempts to verify with it will also fail.

For example, the original public key I'm currently using is:
00: 15 1d af e7 69 47 22 9c 
08: 34 c1 6b 1b ff af 1b f5 
10: 9a e2 ec 61 17 75 35 90 
18: 43 5d 0e 3f 35 56 25 b5 
20: 7c 27 f6 52 82 c5 64 a4 
28: 28 df c1 8d 7f 8c 80 41 
30: 45 33 cf 61 92 92 1d ba 
38: 12 e6 b4 79 f2 29 6b 36 

This compresses to:
00: 02 15 1d af e7 69 47 22 
08: 9c 34 c1 6b 1b ff af 1b 
10: f5 9a e2 ec 61 17 75 35 
18: 90 43 5d 0e 3f 35 56 25 

And then decompresses to:
00: 15 1d af e7 69 47 22 9c 
08: 34 c1 6b 1b ff af 1b f5 
10: 9a e2 ec 61 17 75 35 90 
18: 43 5d 0e 3f 35 56 25 45 
20: 95 76 37 3c 5b 38 6a dd 
28: f8 e8 6a 83 4e 12 1c bb 
30: d5 67 2e d7 61 65 c3 6b 
38: 7a ce 9c 2a bc b8 b4 c8 

At this point attempts to verify with that key will fail.

Should I be able to compress/decompress keys generated with this curve? Thank you for any help!

kmackay, 2015-03-01:
It looks like your buffer for the compressed point is too small. It should be 33 bytes in this case (uECC_BYTES + 1).

totalslacker, 2015-03-01:
Ugh - I’m an idiot!

Thank you!

Shannon

closed 2015-03-01 by totalslacker