kmackay/micro-ecc #24: signature verification fails

anishn63, 2015-03-13:
Hi,
I am testing with OpenSSL ECDSA signature. The test procedure is to create a ECDSA a signature in OpenSSL, then verify this signature using micro-ecc (publickey and hash is taken from OpenSSL output). I found that except for "secp160r1" curve, the signature verification fails in micro-ecc for the other supported curves. Is this expected?

kmackay, 2015-03-13:
You need to compile micro-ecc for the curve you want to use (by setting the
uECC_CURVE #define). The default setting is for secp160r1.

anishn63, 2015-03-14:
Yes, I have already done that ( change uECC_CURVE macro according to the curve).

kmackay, 2015-03-14:
I can't really tell you what the issue is then. It does work correctly with
all supported curves. Make sure that you are using the correct buffer sizes.

Note that you need to set uECC_CURVE either by modifying the uECC.h header
file directly, or by setting it in your Makefile or whatever you are using
(using -DuECC_CURVE=whatever). You can't just set #define uECC_CURVE
secp256r1 (for example) in your own code before including uECC.h; uECC.c
must also be compiled with your desired curve.

closed 2015-03-25 by kmackay

AlphaBravoCharly, 2017-04-04:
Hi kmackay,

Two years passed since you've closed this question, and now it looks like I face a similar challenge... I did compile it with  -DuECC_CURVE=1 and also tried to see how it works on debugger. It obviously does use this curve but verification does not pass.

I presume my problem is just somewhere in the way I feed data from OpenSSL to uECC: I provide hash (SHA256), pubulic key (both BIGNUMs) and signature, but it does not verify :(

May be you could post some testing data that you've used to compare it against OpenSSL for NIST P-265 please? Did you use any specific procedure to take components from DER for example? Anything else that could help? Or, just 1-2 test sets would also be very helpful!

Many many thanks in advance!

arnavblockwala, 2018-03-19:
Hi AlphaBravoCharly,
NIST P-265 is not supported in this library. I think you are confusing it with secp256k1.