kmackay/micro-ecc #26: request for help asap

azahido, 2015-04-06:
Hello, 
please, I come back to you Mr. I lost in optimization to run the mico-ecc in my environment, the issue when I run the ecc- signature generation it took more 19 seconds   please, I lost  more time (more than 4 months) to have results as you found less than 2 seconds. 
I am working on "cortus aps 32 bits as platform". please I want to understand how you got the very good answers time results, I don't know If I need to write the assembly for the function that take time in calculation, if needed please help me.
Please, I would be very grateful if you could help me asap to resolved this issue.
Thank you so much

kmackay, 2015-04-06:
Make sure you are using the correct uECC_WORD_SIZE (I assume 4). Other than
that, you will need to implement operations for your processor to improve
performance. I don't know anything about the Cortus processor. If they are
ARM-based then you can just use the existing ARM assembly implementation
(define uECC_PLATFORM to uECC_arm). Otherwise you will need to implement
the assembly implementation yourself.

azahido, 2015-04-07:
please, when which step I can use the ARM-Assembly could you help me by
explaining the steep to use  existing assembly

kmackay, 2015-04-10:
Apparently Cortus chips are not ARM-based, they use their own instruction set. This means that you will not be able to use my existing ARM assembly implementation. You will need to write your own implementation. I also wasn't able to find any documentation on the Cortus instruction set, so you're on your own there.

I recommend starting with the vli_add() and vli_sub() since they are pretty simple and you should see a measurable performance increase. For fast multiplication (vli_mult() and vli_square()), see http://www.iacr.org/archive/ches2011/69170459/69170459.pdf

closed 2015-04-10 by kmackay