kmackay/micro-ecc #28: asm_avr_fast

azahido, 2015-04-20:
hello again, 
please, I am training to test the project with arduino  leonardo but I got some error related to some files not found:

ecc_test.ino:7:17: fatal error: j0g.h: No such file or directory

please, where can I find those files:
#include <j0g.h>
#include <js0n.h>
#include <lwm.h>
#include <bitlash.h>
#include <GS.h>
#include <Scout.h>
#include <Shell.h>

when disable all the files mentioned  above I got the result without using the asm fast, 
please, could you help me to run the asm fast to get the result like that you found : 

"
ECDSA (fast)

In these tests, uECC_ASM was defined to uECC_asm_fast and ECC_SQUARE_FUNC was defined to 1 in all cases.

secp160r1	secp192r1	secp256r1	secp256k1
ECDSA sign time (ms):	555	902	2386	1773
ECDSA verify time (ms):	590	990	2650	1800
Code size (bytes):	13246	14798	22594	22826
"
thank you so much for all your support 
thant again,

kmackay, 2015-04-22:
I removed the unneeded includes. The speed is mostly dependent on the RNG function. The implementation I provided in ecc_test.ino is fairly good in terms of randomness, but is very slow if there is not a lot of noise on pin 0. I find that it is a lot faster if I hold my hand next to the board, for example. You should either connect pin 0 up to a source of random noise, or implement your own RNG function.

closed 2015-04-22 by kmackay

azahido, 2015-04-22:
Hello,
Hope all is ok for you! thank you again for all your time and support
OK,
please, back to my issue as I told two week ago, I am using the cortus aps
 has not a HW multiplier integrated.
could you have any proposal to help me because the issue from my side I
found that the mico used have not the multiplier hardware integrated
please, if you have any proposal to improve the result know I am still in
13 seconds needed to generate signature using the 160r1, (vli_mod_fast used
or any other ideas)
Thank you so much for all your feedback really I am lost in this uECC
implementation.
have nice day

kmackay, 2015-04-22:
Your Cortus microcontroller has no HW multiply? That will make things slow :-(

Your only option I think is to write your own assembly optimizations. As I said before, I can't find any instruction set reference for the Cortus chips, so I can't really help you.

I recently saw a new paper on speeding up multiplication that may be useful to you: https://cryptojedi.org/papers/avrmul-20140715.pdf
It is AVR-specific, but should be easily adapted to the Cortus chip.

azahido, 2015-04-23:
Hello  Dear,
Thank you so much for all your help,  as I told you the cortus doasn't have
a hw multiplier,
so please, could you help me to implement the curve 160k1  I want to used
it instead of 160r1  As i understand the koblitz is faster than random
curve is it?
please could you share with me the code for 160k1 if you have it if not
could you help me to do it!

Thank a lot for your understanding   and for all your time

azahido, 2015-04-23:
Hello Dear,
please, do you have any implementation about ecc over binary field I want
to test with it.
Thanks a lot.

kmackay, 2015-04-24:
I don't have any implementation of ECC over a binary field. I also don't
have an implementation of secp160k1; I expect that it would be about 30%
faster than secp160r1 (going by the difference between secp256k1 and
secp256r1). I don't have any plans to implement secp160k1, but it should be
fairly easy for you to implement it based on the secp256k1 code.

azahido, 2015-04-24:
please, If i understaind the difference betwen the secp160k1 and secp256k1
is in the function omega_mult( uint32_t* RESTRICT p_result, uint32_t*
RESTRICT p_right ) please, I saw some comment but I didn't found any
article that expline why you multiple by:
     /* Multiply by (2^9 + 2^8 + 2^7 + 2^6 + 2^4 + 1). */

 for ( k = 0; k < uECC_WORDS; ++k )
    {
        uint64_t p = ( uint64_t )0x3D1 * p_right[k] + l_carry;
        p_result[k] = ( p & 0xffffffff );
        l_carry = p >> 32;
    }
and add:
 /* add the 2^32 multiple */
    p_result[1 + uECC_WORDS] = vli_add( p_result + 1, p_result + 1, p_right
); /* add the 2^32 multiple */

please, I sorry to disturb but I lost I want to have some result bu this
curve  secp160k1  if it can give a result les than 30% will be perfect
please could you help me to implement this function to have good result if
it possible  I would be very grateful if you could help me with this matter.

Thanks again,

kmackay, 2015-04-25:
secp256k1 is defined over a prime field with p = 2^256 - 2^32 - 2^9 - 2^8 - 2^7 - 2^6 - 2^4 - 1. With primes of this form there is a faster algorithm for reduction modulo p for numbers less than about p^2 (ie, it works for products of two field elements); see http://www.isys.uni-klu.ac.at/PDF/2001-0126-MT.pdf page 354 (Algorithm 4).

You can use the same algorithm for secp160k1, but the polynomial is 2^160 - 2^32 - 2^14 - 2^12 - 2^9 - 2^7 - 2^3 - 2^2 - 1, so your omega_mult() function would multiply by 2^32 - 2^14 - 2^12 - 2^9 - 2^7 - 2^3 - 2^2 - 1 instead.

Note that you will also need to define the correct values for Curve_P, Curve_B, Curve_G, and Curve_N. See http://www.secg.org/SEC2-Ver-1.0.pdf for the correct values.

azahido, 2015-04-27:
hello,
please, I have tired to found where the issue come from but no way, I
followed your step and the code is in attached with 160k1 included but the
result not correct I compare the result with a tool SCV cryptomanager as I
did for all curve to validate result please, if possible to take a look and
tell me where wrong result come from, I am so sorry to disturb you but I
find you are  the best one that can help me to overcome this  real problem
from my side thanks again.
Thanks a lot ,
Best regardes

azahido, 2015-04-27:
Thank you so much I resolved the problem thanks a lot

azahido, 2015-04-27:
But the resuult  of 160r1 is better than 160k1  i did not know why the last
time you told me that the of 160k1 should be better than 160r1 with ratio
30% but is not the case could you confirm please

kmackay, 2015-04-28:
That was just based on the difference between secp256r1 and secp256k1. I don't know what the difference between secp160r1 and secp160k1 should be. Make sure you have set uECC_WORDS to the correct value (5).

I highly recommend that you implement asm optimizations since it will give you a significant speed improvement. For example on the AVR the inline asm version is more than 5 times faster than the plain C version.

azahido, 2015-04-28:
please could you tell me if you have an algorithm used to optimize the asm
or generate by an external compiler I don't know could you tell me about
the procedure used if you want I will send you the coud generated by aps
compiler if you can help and also the datasheet included the instruction
asm.
please,I am really stuck and it can cause to lose my job I am working on
that more 5 months or more as I told you 1 month ago.
Thanks for all your support

azahido, 2015-04-28:
please, another request I need to try with ECC over binary field instead of
F(p), beause my hardware has not an hardware multiplier I see the binary
field will be batter than prime field,
please, where I can find an implementation with c code. help me please,

azahido, 2015-04-28:
please, another question if it possible to use your implementation (uECC)
implementated over  F(p)  prime field and use it for f(2^m) binary field by
make some change in the algo what do you think please,