kmackay/micro-ecc #30: using a public/private key typedef instead of uint8_t key[ECC_BYTES*n]

jnysen, 2015-04-25:
One useful thing to help debug and pin point issues in new implementations is the use of a specific typedef to differentiate the different byte strings that are being passed in to/out of the interface function calls. This allows many issues to be found at compile time.

eg. 

instead of this:
func( const uint8_t public_key[ECC_BYTES*2], const uint8_t private_key[ECC_BYTES], uint8_t signature[ECC_BYTES*2] );

this:
typedef uint8_t ECC_PrivateKey_t[ECC_BYTES];
typedef uint8_t ECC_Signature_t[ECC_BYTES*2];
typedef uint8_t ECC_PublicKey_t[ECC_BYTES*2];
typedef uint8_t ECC_CompressedPublicKey_t[1+ECC_BYTES];

func( ECC_PublicKey_t* const p_public_key, ECC_PrivateKey_t* const p_private_key, ECC_Signature_t*  p_signature );

Obviously this means lots of typecasting or a typecast intermediary local var within uECC. And it means existing implementations need to change their interface vars, but IMO a type defined external interface would assist newcomers with finding problems where they've transposed a public key, private key, or signature.

Not a real problem, as it was simple enough to add inlined casting wrappers to translate the functional interface differences without impacting the final compiled code.

eg.
inline ECC_BoolRet_t x_uECC_MakeKey( ECC_PublicKey_t *pxPublicKey, ECC_PrivateKey_t *pxPrivateKey )
{
  return (ECC_BoolRet_t) uECC_make_key( (uint8_t *)pxPublicKey, (uint8_t *)pxPrivateKey );
}

kmackay, 2015-04-25:
I prefer not to do that in uECC, but as you said you are welcome to create your own wrappers.

closed 2015-04-25 by kmackay

jnysen, 2015-04-28:
I understand completely. One of the reasons I like uECC is that the code is clean and concise without having to drive up and down through numerous layers of abstraction to see how it works. Keeping it simple also makes platform ports and language wrappers far easier to write.

For those interested, the following is a snippet of my C language interface code that provides an emulation of strong type checking at compile time:

```C
// ----------------------------------------------------------------------------- //
// uECC wrappers

typedef struct
{
	uint8_t uc[uECC_BYTES];
} ECC_PrivateKey_t;

typedef struct
{
	uint8_t uc[uECC_BYTES*2];
} ECC_PublicKey_t;

typedef struct
{
	uint8_t uc[uECC_BYTES+1];
} ECC_CompressedPublicKey_t;

typedef struct
{
	uint8_t uc[uECC_BYTES];
} ECC_Block_t;

typedef struct
{
	uint8_t uc[uECC_BYTES*2];
} ECC_Signature_t;

inline int xECC_MakeKeyPair( ECC_PublicKey_t *pxPublicKey, ECC_PrivateKey_t *pxPrivateKey )
{
	return uECC_make_key( (uint8_t *)pxPublicKey, (uint8_t *)pxPrivateKey );
}

inline int xECC_CalculateSharedSecret( const ECC_PublicKey_t *pxPublicKey, const ECC_PrivateKey_t *pxPrivateKey, ECC_Block_t *pxSecret )
{
	return uECC_shared_secret( (uint8_t *)pxPublicKey, (uint8_t *)pxPrivateKey, (uint8_t *)pxSecret );
}

inline int xECC_ValidatePublicKey( const ECC_PublicKey_t *pxPublicKey )
{
	return uECC_valid_public_key( (uint8_t *)pxPublicKey );
}

inline void vECC_Compress( const ECC_PublicKey_t *pxPublicKey, ECC_CompressedPublicKey_t *pxCompressedPublicKey )
{
	uECC_compress( (uint8_t *)pxPublicKey, (uint8_t *)pxCompressedPublicKey );
}

inline void vECC_Decompress( const ECC_CompressedPublicKey_t *pxCompressedPublicKey, ECC_PublicKey_t *pxPublicKey )
{
	uECC_decompress( (uint8_t *)pxCompressedPublicKey, (uint8_t *)pxPublicKey );
}

inline int xECC_Sign( const ECC_PrivateKey_t *pxPrivateKey, const ECC_Block_t *pxHash, ECC_Signature_t *pxSignature )
{
	return uECC_sign( (uint8_t *)pxPrivateKey, (uint8_t *)pxHash, (uint8_t *)pxSignature );
}

inline int xECC_Verify( const ECC_PublicKey_t *pxPublicKey, const ECC_Block_t *pxHash, const ECC_Signature_t *pxSignature )
{
	return uECC_verify( (uint8_t *)pxPublicKey, (uint8_t *)pxHash, (uint8_t *)pxSignature );
}
```