kmackay/micro-ecc #31: EccPoint_mult and point addition

callalilychen, 2015-04-29:
Hi,

I'd like to use your uECC library for my master thesis. I need a function to do point addition on a curve (aG+bG, a and b are integers). So I wrote the following code at the end of uECC.c:

```
typedef struct key_pair{
  uECC_word_t d[uECC_BYTES];
  EccPoint Q;
} key_pair;

void print_bytearray(uint8_t *bytes, int length){
  int i = 0;
  for(i = 0; i < length; i++){
    printf("%x ", *(bytes+i));
  }

}
void print_point(EccPoint point){
  printf("(\n");
  print_bytearray(point.x, sizeof(point.x));
  printf(",\n");
  print_bytearray(point.y, sizeof(point.y));
  printf(")\n");
}

 void EccPoint_add(EccPoint *  p_P, EccPoint *  p_Q, EccPoint *  p_result){
    uECC_word_t P_x[uECC_WORDS];
    uECC_word_t P_y[uECC_WORDS];
    uECC_word_t Q_x[uECC_WORDS];
    uECC_word_t Q_y[uECC_WORDS];
    
    vli_set(P_x, p_P->x);
    vli_set(P_y, p_P->y);
    vli_set(Q_x, p_Q->x);
    vli_set(Q_y, p_Q->y);
    
    XYcZ_addC(P_x, P_y, Q_x, Q_y);

    vli_set(p_result->x, Q_x);
    vli_set(p_result->y, Q_y);
}

int main(){
  key_pair k_pair, test_pair;
  
  uECC_make_key(&k_pair.Q, k_pair.d);
  uECC_make_key(&test_pair.Q, test_pair.d);

  EccPoint p1, p2;

  EccPoint_add(&k_pair.Q, &test_pair.Q, &p1);
  printf("Get p1:\n");
  print_point(p1);

  uECC_word_t sum[uECC_BYTES];
  vli_modAdd(sum, k_pair.d, test_pair.d, curve_n);
  EccPoint_mult(&p2, &curve_G, sum, 0, vli_numBits(sum, uECC_WORDS));
  printf("Get P2:\n");
  print_point(p2);
  // FIXME p1 != p2

  return 1;
}
```

But I got two different points, when I compute aG+bG and (a+b)G. It will be very nice, if you can help me to debug my *EccPoint_add* function.  Can you please tell me, if I have used the *XYcZ_addC* function there correctly, or if there is any better function for the point addition? Can you also explain me about the variable *p_initialZ* in the *EccPoint_mult* function. I'm sorry, but I do not understand the z value, which you have also written in comments.

Best regards,

Wenwen

jnysen, 2015-04-29:
Make sure you set your random function before calling make key.

kmackay, 2015-04-30:
jnysen is correct that you need to call uECC_set_rng() first if you are running on an embedded platform. It is not required when running on Windows, OS X, or Linux.

There are a number of other issues with your code:
 * You appear to be compiling without any warnings or error checking, because you are passing the wrong pointer types in several places. Particularly, the values passed to uECC_make_key().
 * If you are using the secp160r1 curve, note that n is larger (more words) than p, so you need to use uECC_N_WORDS and vli_modAdd_n().
 * Your EccPoint_add function is incorrect. uECC uses projective Jacobian coordinates internally for point representation, but the Z coordinate is implicit during the Montgomery ladder (eg in XYcZ_add) since P and Q share the same Z coordinate. See http://eprint.iacr.org/2011/338.pdf for details of this algorithm. The upshot is that after you call XYcZ_add(), you need to find the implicit Z coordinate and then use it to convert from Jacobian coords back to affine coords.
 * Note that you can just use XYcZ_add() rather than XYcZ_addC() for a performance improvement (and it makes it easier to find Z).

Here is the corrected code which works for all curves (I only did minimal testing):

```
typedef struct key_pair {
  uECC_word_t d[uECC_N_WORDS];
  EccPoint Q;
} key_pair;

void print_bytearray(uint8_t *bytes, int length) {
    int i = 0;
    for (i = 0; i < length; i++) {
        printf("%02x ", bytes[i]);
    }
}

void print_point(EccPoint point){
    printf("( ");
    print_bytearray((uint8_t *)point.x, sizeof(point.x));
    printf(",\n  ");
    print_bytearray((uint8_t *)point.y, sizeof(point.y));
    printf(")\n");
}

 void EccPoint_add(EccPoint *  p_P, EccPoint *  p_Q, EccPoint *  p_result){
    uECC_word_t P_x[uECC_WORDS];
    uECC_word_t P_y[uECC_WORDS];
    uECC_word_t Q_x[uECC_WORDS];
    uECC_word_t Q_y[uECC_WORDS];

    vli_set(P_x, p_P->x);
    vli_set(P_y, p_P->y);
    vli_set(Q_x, p_Q->x);
    vli_set(Q_y, p_Q->y);

    XYcZ_add(P_x, P_y, Q_x, Q_y);
    
    /* Find final 1/Z value. */
    uECC_word_t z[uECC_WORDS];
    vli_modMult_fast(z, p_P->x, P_y);
    vli_modInv(z, z, curve_p);
    vli_modMult_fast(z, z, P_x);
    vli_modMult_fast(z, z, p_P->y);
    /* End 1/Z calculation */
    
    apply_z(Q_x, Q_y, z);

    vli_set(p_result->x, Q_x);
    vli_set(p_result->y, Q_y);
}

void make_key_helper(key_pair *pair) {
    memset(pair->d, 0, sizeof(pair->d));
    
    uint8_t publicKey[uECC_BYTES*2];
    uint8_t privateKey[uECC_BYTES];
    uECC_make_key(publicKey, privateKey);
    
    vli_bytesToNative(pair->Q.x, publicKey);
    vli_bytesToNative(pair->Q.y, publicKey + uECC_BYTES);
    vli_bytesToNative(pair->d, privateKey);
}

int main(){
    key_pair k_pair, test_pair;
    
    make_key_helper(&k_pair);
    make_key_helper(&test_pair);
    
    EccPoint p1, p2;
    
    EccPoint_add(&k_pair.Q, &test_pair.Q, &p1);
    printf("Get p1:\n");
    print_point(p1);
    
    uECC_word_t sum[uECC_N_WORDS];
    vli_modAdd_n(sum, k_pair.d, test_pair.d, curve_n);
    EccPoint_mult(&p2, &curve_G, sum, 0, vli_numBits(sum, uECC_N_WORDS));
    printf("Get P2:\n");
    print_point(p2);
    
    return 1;
}
```

callalilychen, 2015-04-30:
Thank you very much for you help and the paper. Now I can just test the code on PC. I will try to understand your code in more details.

closed 2015-04-30 by callalilychen