kmackay/micro-ecc #34: uECC_RNG_Function parameter 2 is "unsigned" without further qualification

jnysen, 2015-05-08:
Using "unsigned" without any further type qualifiers implies 'unsigned short'. This should be explicitly defined with something more appropriate like uint8_t or uint16_t (and preferably the platform natural word size) instead of just "unsigned".

kmackay, 2015-05-08:
"unsigned" is equivalent to "unsigned int", which will often be the natural word size (on 16 and 32-bit platforms). I don't see any reason to specify an explicit size for the type in this case; can you explain your reasoning further?

jnysen, 2015-05-11:
Sorry, this was a platform dependent cross compiler issue with a compiler warning generated when passing a uint32_t function to the uECC_set_rng function. The compiler throws a warning because "unsigned int" is not the same as any of the <stdint.h> uintXX_t types (even though it is equivalent to uint32_t on the ARM platform).

The 'short vs int' issue came up because unsigned short on one processor is equal to unsigned int on the other (where int and short are equivalent, but for some reason are not treated as identical). A pre-processor lexical checker didn't handle the "unsigned" in the system specific RandomFunc() and caused the compile to fail with an error. (This checker has evolved in an automotive build environment where pretty much everything follows the MISRA C guidelines.)

You can close this issue as it appears to be a system specific build problem otherwise unrelated to uECC.

closed 2015-05-16 by kmackay