kmackay/micro-ecc #35: Unknown register name in asm asm_arm.inc

Gavintc, 2015-05-15:
I try to use this library in a iPhone app. After I included uECC.c, uECC.h and asm_arm.inc into my project folder, I fail to build the project due to "Unknown register name in asm_arm.inc". The architecture of iOS app is arm64, armv7 and armv7s. So did I include a wrong .inc file? Thanks.

kmackay, 2015-05-16:
If you are building for arm64, there are no asm optimizations currently. You should define uECC_ASM to uECC_asm_none.

I will attempt to fix the automatic checking for this case.

kmackay, 2015-05-16:
I can't reproduce this issue. Can you tell me which version of Xcode you are using, and how you have configured your project / build settings?

Gavintc, 2015-05-18:
Thanks kmackay. I'm using Xcode 6.3.1 on OS X Yosemite. I just download and import uECC.h, uECC.c and asm_arm.inc into my project. The Architecture of Build Settings is:
Architectures: Standard architectures (armv7, arm64)
Valid Architectures: arm64 armv7 armv7s

While, I just follow your advice and change "uECC_ASM to uECC_asm_none" in uECC.h, and then it won't throw any errors. 
By the way, I just want to do a signature verification of secp256k1 with uECC_verify. The documents say I need to compile with "-DuECC_CURVE=uECC_secp256k1", but how can I set this in Xcode? Notes that there are other C code in my project. Any ideas of how to do a secp256k1 signature verification?

kmackay, 2015-05-18:
The easiest way is to just modify uECC.h: change the line "#define uECC_CURVE uECC_secp160r1" to "#define uECC_CURVE uECC_secp256k1"

Another way to do it is to modify the preprocessor macro settings in your project build settings. I don't really use Xcode much, but here is how I did it: Go to the project build settings (go to the project settings by clicking on the project, and then go to the "Build Settings" tab). Then make sure it is showing "All" (not "Basic"). Then go to the section "Preprocessor Macros" (you can use the search bar). And add a new entry "uECC_CURVE=uECC_secp256k1" in both the debug and release sections.

Gavintc, 2015-05-18:
Great! Thank you so much! The easiest way works for me. By the way, my signature length is 71 bytes while uECC_verify only accepts signature length with 64 bytes. I can't find any functions to transfer the signature. Any idea about how to do it?

kmackay, 2015-05-18:
I'm guessing that you are doing something Bitcoin-related, in which case your signature is DER encoded. You need to decode it to get the raw signature to pass in to uECC_verify(). See [here](https://github.com/bitcoin/bitcoin/blob/ef9acc5124459b111d8643323a71d8198d0eaae8/src/script.cpp#L245)

Gavintc, 2015-05-19:
Yes, my signature is DER decoded. And for your reference, this is the raw bytes of my signature,
3045 0221 0088 9a34 54e7 f4f8 eee6 d984
a00f 73f7 d564 4d92 0e49 526c 1c30 6074
2283 1bd1 5a02 2000 c03e a953 89f8 5393
4730 4dc7 7b44 b3a7 cf4b 8114 f569 431e
6de4 d591 8a64 8b
I found that in my signature, length of R is 33 bytes and length of S is 32 bytes. They sum up to 65 bytes, still not 64 bytes. So how to decode it to 64 bytes signature correctly?

kmackay, 2015-05-19:
The reason R is 33 bytes is because the highest bit is set, so there is a 0 byte prepended (to make the overall integer "positive" in a signed representation. To decode, just skip the 0 byte; in your example, I think R would start at the 0x88 byte.

closed 2015-05-26 by kmackay