kmackay/micro-ecc #43: Running the test code

Vijayjatti, 2015-06-23:
Hello,
              I am a newbie to ECC library. I was trying with the micro-ecc code, but after trying for a long time I was unable to run it in Visual Studio. When I run the test_ecdh.c the command prompt appears with a message "Testing 256 random private key pairs" and closes after some time. I think the issue may be with LPC11XX but not sure. Kindly help me to resolve the issue.

Thank you in advance and sorry for asking such a stupid question.

Thank you
Vijay

kmackay, 2015-06-24:
I don’t have immediate access to Visual Studio, so it may take me some time to determine what is going on. The last time I tried it in VS was over a year ago, so it’s possible that one of the more recent changes broke it. You could try adding additional printing to test_ecdh.c or uECC.c to determine if it is hanging somewhere (or use the debugger).

Vijayjatti, 2015-06-24:
Thank you for your reply . I am really keen to work on ECDH. If possible kindly provide me any tutorials so that I can learn more about it. It would be a great help.

Thank you
Vijay

kmackay, 2015-06-24:
I don’t really have any tutorials; the Wikipedia page is probably a good place to start….

For test_ecdh.c, when it is working correctly it just prints out 256 dots (.). So, maybe it is working fine and you just didn’t notice the output? It only prints something (other than the initial “Testing 256 random private key pairs”) if an error ocurrs.

Vijayjatti, 2015-06-24:
Thank you. Your replies really helped me a lot.

Thank you
Vijay

Vijayjatti, 2015-06-25:
Hello,       Sorry to bother you again. When I compile the same program by changing it to secp256r1in uECC 
/* Curve selection options. */#define uECC_secp160r1 1#define uECC_secp192r1 2#define uECC_secp256r1 3#define uECC_secp256k1 4#ifndef uECC_CURVE    #define uECC_CURVE uECC_secp256r1#endif
it shows me lots of errors in uECC.c that I am not able to understand. Could you please help me

Regards,
Vijaykumar JattiM.S Computer Science and EngineeringSanta Clara UniversitySanta Clara, California

From: vijay_jatti@hotmail.com
Subject: Re: [micro-ecc] Running the test code (#43)
Date: Tue, 23 Jun 2015 21:36:56 -0700
To: reply@reply.github.com

Thank you. Your replies really helped me a lot.
Thank youVijay

kmackay, 2015-06-25:
Maybe you could paste in the errors?

Vijayjatti, 2015-06-25:
1>------ Build started: Project: ecdh, Configuration: Debug Win32 ------1>  uECC.c1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2109): error C2275: 'bitcount_t' : illegal use of this type as an expression1>          c:\users\vijay\desktop\micro-ecc-master\uecc.c(185) : see declaration of 'bitcount_t'1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2109): error C2146: syntax error : missing ';' before identifier 'i'1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2109): error C2065: 'i' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2110): error C2275: 'uECC_word_t' : illegal use of this type as an expression1>          c:\users\vijay\desktop\micro-ecc-master\uecc.c(181) : see declaration of 'uECC_word_t'1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2110): error C2146: syntax error : missing ';' before identifier 'index'1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2110): error C2065: 'index' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2111): error C2065: 'i' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2112): error C2065: 'index' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2113): error C2065: 'index' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2116): error C2065: 'index' : undeclared identifier1>  test_ecdh.c1>  Generating Code...========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

Regards,
Vijaykumar JattiM.S Computer Science and EngineeringSanta Clara UniversitySanta Clara, California

Date: Wed, 24 Jun 2015 18:03:26 -0700

Vijayjatti, 2015-06-25:
Hello, These are the changes that I have made. I want to pass these strings as shown in changes_test_ecdh for testing and get the output. But I have no clue what exactly I have to do with the present code. Just if possible kindly give me a road map to achieve it. I am stuck and clueless. 

Thank you,Vijay
From: vijay_jatti@hotmail.com
To: reply@reply.github.com
Subject: RE: [micro-ecc] Running the test code (#43)
Date: Wed, 24 Jun 2015 18:04:58 -0700

1>------ Build started: Project: ecdh, Configuration: Debug Win32 ------1>  uECC.c1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2109): error C2275: 'bitcount_t' : illegal use of this type as an expression1>          c:\users\vijay\desktop\micro-ecc-master\uecc.c(185) : see declaration of 'bitcount_t'1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2109): error C2146: syntax error : missing ';' before identifier 'i'1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2109): error C2065: 'i' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2110): error C2275: 'uECC_word_t' : illegal use of this type as an expression1>          c:\users\vijay\desktop\micro-ecc-master\uecc.c(181) : see declaration of 'uECC_word_t'1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2110): error C2146: syntax error : missing ';' before identifier 'index'1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2110): error C2065: 'index' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2111): error C2065: 'i' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2112): error C2065: 'index' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2113): error C2065: 'index' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2116): error C2065: 'index' : undeclared identifier1>  test_ecdh.c1>  Generating Code...========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

Regards,
Vijaykumar JattiM.S Computer Science and EngineeringSanta Clara UniversitySanta Clara, California

Date: Wed, 24 Jun 2015 18:03:26 -0700

Vijayjatti, 2015-06-25:
I want to achieve as it is being given inhttps://github.com/IOT-DSA/docs/wiki/Connection-Test-Case
I am sorry for thread of messages. But really need your help how I can go with it. Any suggestion would be a great help for me

From: vijay_jatti@hotmail.com
To: reply@reply.github.com
Subject: RE: [micro-ecc] Running the test code (#43)
Date: Wed, 24 Jun 2015 18:13:52 -0700

Hello, These are the changes that I have made. I want to pass these strings as shown in changes_test_ecdh for testing and get the output. But I have no clue what exactly I have to do with the present code. Just if possible kindly give me a road map to achieve it. I am stuck and clueless. 

Thank you,Vijay
From: vijay_jatti@hotmail.com
To: reply@reply.github.com
Subject: RE: [micro-ecc] Running the test code (#43)
Date: Wed, 24 Jun 2015 18:04:58 -0700

1>------ Build started: Project: ecdh, Configuration: Debug Win32 ------1>  uECC.c1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2109): error C2275: 'bitcount_t' : illegal use of this type as an expression1>          c:\users\vijay\desktop\micro-ecc-master\uecc.c(185) : see declaration of 'bitcount_t'1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2109): error C2146: syntax error : missing ';' before identifier 'i'1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2109): error C2065: 'i' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2110): error C2275: 'uECC_word_t' : illegal use of this type as an expression1>          c:\users\vijay\desktop\micro-ecc-master\uecc.c(181) : see declaration of 'uECC_word_t'1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2110): error C2146: syntax error : missing ';' before identifier 'index'1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2110): error C2065: 'index' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2111): error C2065: 'i' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2112): error C2065: 'index' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2113): error C2065: 'index' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2116): error C2065: 'index' : undeclared identifier1>  test_ecdh.c1>  Generating Code...========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

Regards,
Vijaykumar JattiM.S Computer Science and EngineeringSanta Clara UniversitySanta Clara, California

Date: Wed, 24 Jun 2015 18:03:26 -0700

kmackay, 2015-06-25:
Oh, it looks like you are compiling it as C. Unfortunately Visual Studio doesn’t really support C (at least c99) which is why you are seeing these errors. I can probably fix it, but you might want to try just compiling it as c++.

> On Jun 24, 2015, at 6:13 PM, Vijaykumar Jatti <notifications@github.com> wrote:
> 
> Hello, These are the changes that I have made. I want to pass these strings as shown in changes_test_ecdh for testing and get the output. But I have no clue what exactly I have to do with the present code. Just if possible kindly give me a road map to achieve it. I am stuck and clueless. 
> 
> Thank you,Vijay
> From: vijay_jatti@hotmail.com
> To: reply@reply.github.com
> Subject: RE: [micro-ecc] Running the test code (#43)
> Date: Wed, 24 Jun 2015 18:04:58 -0700
> 
> 
> 
> 
> 1>------ Build started: Project: ecdh, Configuration: Debug Win32 ------1> uECC.c1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2109): error C2275: 'bitcount_t' : illegal use of this type as an expression1> c:\users\vijay\desktop\micro-ecc-master\uecc.c(185) : see declaration of 'bitcount_t'1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2109): error C2146: syntax error : missing ';' before identifier 'i'1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2109): error C2065: 'i' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2110): error C2275: 'uECC_word_t' : illegal use of this type as an expression1> c:\users\vijay\desktop\micro-ecc-master\uecc.c(181) : see declaration of 'uECC_word_t'1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2110): error C2146: syntax error : missing ';' before identifier 'index'1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2110): error C2065: 'index' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2111): error C2065: 'i' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2112): error C2065: 'index' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2113): error C2065: 'index' : undeclared identifier1>c:\users\vijay\desktop\micro-ecc-master\uecc.c(2116): error C2065: 'index' : undeclared identifier1> test_ecdh.c1> Generating Code...========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========
> 
> 
> Regards,
> Vijaykumar JattiM.S Computer Science and EngineeringSanta Clara UniversitySanta Clara, California
> 
> Date: Wed, 24 Jun 2015 18:03:26 -0700
> From: notifications@github.com
> To: micro-ecc@noreply.github.com
> CC: vijay_jatti@hotmail.com
> Subject: Re: [micro-ecc] Running the test code (#43)
> 
> Maybe you could paste in the errors?
> 
> 
> 
> > On Jun 24, 2015, at 5:01 PM, Vijaykumar Jatti <notifications@github.com> wrote:
> 
> > 
> 
> > Hello, Sorry to bother you again. When I compile the same program by changing it to secp256r1in uECC 
> 
> > /* Curve selection options. */#define uECC_secp160r1 1#define uECC_secp192r1 2#define uECC_secp256r1 3#define uECC_secp256k1 4#ifndef uECC_CURVE #define uECC_CURVE uECC_secp256r1#endif
> 
> > it shows me lots of errors in uECC.c that I am not able to understand. Could you please help me
>

Vijayjatti, 2015-06-25:
Thank you so much 😊

kmackay, 2015-06-25:
Looking at https://github.com/IOT-DSA/docs/wiki/Connection-Test-Case, I don't think any modifications are required to micro-ecc. Instead, you should write some wrapper code that uses micro-ecc to dodo ECDH with the given keys. You will need to use / write something to do base64 encode/decode and SHA256.

Vijayjatti, 2015-06-26:
I am making use of test_ecdh to achieve ECDH. Just wanted to confirm if I am using the right code to do ecdh

kmackay, 2015-06-26:
To do ECDH, use uECC_shared_secret()

closed 2015-07-11 by kmackay