kmackay/micro-ecc #49: Fail at repeated key generation.

Xywzel, 2015-07-20:
I'm using this test power of different signature methods and for some reason the creation of 1020th signature fails every time, while 1019 before it work fine. What are the reasons for signature failure?

kmackay, 2015-07-21:
Do you mean that uECC_sign() returns 0? This happens if a "bad" random number is generated too many times in a row (currently 64). If your random number generator is correct (ie, cryptographically strong) this is extremely unlikely, and would not happen repeatably.

Can you give me more information about your test? Do you have the test source code available?

closed 2015-08-08 by kmackay

BLOondy, 2017-06-29:
Hi! I have the same problem when I sign the first time it's works but when I do it a second time it's block on uEcc _Sign 2 minute and it begin the program to the very begining.

I have a look and it's on the function uECC_generate_random_int where it's stop. 

What can be the problem ? 

THank you !

kmackay, 2017-06-29:
What platform are you running on? Did you set the RNG function using uECC_set_rng()?