kmackay/micro-ecc #50: Possibility to use more than one curve at once

juliabory, 2015-07-22:
I have a feature request. Is it possible that you change your library in the way that more than one curve can be used simultaneous. We have an application using different curves in runtime and would love to use your lib furthermore. I would be very thankful if you could make a statement about this, so that we could plan our project. Thank your very much!!

quartzjer, 2015-07-22:
I'll probably need this at some point in the near future too, my temporary plan was just to have a script that batch renames things so that there are different symbols for each curve, seemed like an easy enough compromise but haven't tried it yet :)

kmackay, 2015-07-23:
I am working on it. The eventual plan is to have a single #define that controls the size/speed tradeoff, and #defines to determine which curves are supported.

juliabory, 2015-07-23:
Sounds good! I'm looking forward your solution! Thank your very much!

kmackay, 2015-10-13:
master branch now supports runtime curve selection. I still need to add fast mult/square asm for AVR, and also implement optimization level 4 for both platforms.

juliabory, 2016-01-07:
We integrated the new version in december. Very good. Thank you for that! Everything is working well. Today I noticed that our communication is about 250ms slower now. We have a long time graph of the duration in communication and after my checkin in december we have that 250ms jump in the graph. I set uECC_OPTIMIZATION_LEVEL=4 now but no change. Are there any new options I have to use? What is the new VLI thing?

kmackay, 2016-01-08:
Which platform are you using (and which curves)? Unfortunately I've been pretty busy for a while, so haven't had time to work on improving the optimizations.

juliabory, 2016-01-08:
It's an ARM Cortex M0. The tests running with secp192r1. Unfortunately I have no measurements for the secp256r1 but we also use it.

kmackay, 2016-01-09:
Oh, ok. Adding the size parameter probably hurts Cortex M0 more because it has so few useful registers. I imagine that optimization level 4 will help, when I implement it. I am also looking into some improved multiplication/squaring implementations for Cortex M0.

juliabory, 2016-01-11:
That would be nice! Please keep me updated. Thank you very much for that nice library.

juliabory, 2016-01-14:
Should we discuss this in a new issue? I integrated your newest checkin. See the graphic for performance. It is not much better. We use uECC_OPTIMIZATION_LEVEL=4 is this right or should I change it?
[image]

closed 2016-02-22 by juliabory