kmackay/micro-ecc #59: Compilation warning and error

vahidshirvani, 2015-10-23:
I'm trying to compile this for a nRF51 chip (contains a Cortex-M0). If I set these flags `-O0 -fomit-frame-pointer` then it only gives me warning
```
Compiling file: uECC.c
C:\Users\vahid\AppData\Local\Temp\ccL0nRJa.s: Assembler messages:
C:\Users\vahid\AppData\Local\Temp\ccL0nRJa.s:279: Warning: register range not in ascending order
C:\Users\vahid\AppData\Local\Temp\ccL0nRJa.s:279: Warning: register range not in ascending order
C:\Users\vahid\AppData\Local\Temp\ccL0nRJa.s:301: Warning: register range not in ascending order
```
But if I set the flag to `-O`, `-O1` or `-O2` then it results in an error.
```
Compiling file: uECC.c
C:\Users\vahid\AppData\Local\Temp\ccPQcEDS.s: Assembler messages:
C:\Users\vahid\AppData\Local\Temp\ccPQcEDS.s:121: Error: invalid immediate for address calculation (value = 0x0000000A)
make[1]: *** [_build/uECC.o] Fel 1
```
(`Fel` means error in Swedish)
The flag `-O3` generates many errors of the above type.
By the way I sat the flags `-O0 -S` and then looked inside the generated uECC.o file. One of the push instructions on row 279 looked like this
```
3: 
push {r3, r7, r6, r5} 
```
I think this is the cause of the warning above.

kmackay, 2015-10-25:
Should be fixed now.

vahidshirvani, 2015-10-26:
Sorry for the late answer. I did not have any access to my computer during the weekend. I just compiled the code and it didn't give me any warning nor error. I tried with different flags, all of them work now. You are a lifesaver.

closed 2015-10-26 by vahidshirvani