kmackay/micro-ecc #63: Many warnings like "discard const qualifier" and an error "previous declaration"

juliabory, 2016-01-14:
Since the new checkin I get warnings and an error. I can fix it with compiler flag I think but I wanted you to know.

uECC.c:1242:29: warning: passing argument 1 of 'hash_context->init_hash' discards 'const' qualifier from pointer target type
...
uECC.h:299:5: note: previous declaration of 'uECC_sign_deterministic' was here

kmackay, 2016-01-15:
OK. I will fix it when I get back from vacation.

kmackay, 2016-01-25:
I can't reproduce the errors you mention. Perhaps your uECC.h is out of date somehow? In the latest version, hash_context->init_hash() takes a "const struct uECC_HashContext *context" as the first argument anyway, so this error shouldn't occur.

juliabory, 2016-02-10:
OK no problem, maybe you are right. The project is running with the old version now.

closed 2016-02-10 by juliabory