kmackay/micro-ecc #65: API change to capture "recovery id" during signing

peter-conalgo, 2016-01-23:
I want to use your wonderful library for message signing in Bitcoin. However, the standard bitcoin signature format includes 2 extra bits: "recid" or "recovery id". This value captures which of the four possible X values that (R, S) imply and allow the specific public key to be recovered from the signature itself. The process is [nicely described in this answer on stackexchange](http://bitcoin.stackexchange.com/a/38909).

The problem is `uECC_sign_with_k` does the right math, and makes valid signatures, but the values I need for `recid` are not exposed.

I am tempted to change/extend your code to capture these values during the signing process. However, it's a bit beyond my pay grade. What are your thoughts?

kmackay, 2016-01-25:
Looks like you should be able to just use:

recid = p[curve->num_bytes] & 0x01

at uECC.c:1200 (in the current revision).

Technically you also need to account for overflow mod n, but that will basically never happen.

peter-conalgo, 2016-01-25:
Thanks! Do you want a pull request which does that? The problem is I have to change the API.

kmackay, 2016-01-25:
Try it out and see if it works for you locally first. I'll think about whether/how to do an API change.

peter-conalgo, 2016-01-25:
Hmm. Didn't work. Still gets wrong answer about 50% of the time. I wonder if the "side channel" protection stuff is a factor here.

kmackay, 2016-01-25:
OK, I'll look into it in more detail

kmackay, 2016-01-29:
My mistake, it should be:

recid = p[num_words] & 0x01;

peter-conalgo, 2016-01-29:
Yes. Works great! I should have noticed that.

mengguang, 2019-04-05:
+1 for this feature. It's great to have recovery ID.

Arachnid, 2022-01-06:
It would be great to see this small feature integrated into the API!

JiapengLi, 2023-03-31:
For those who need recid and pubkey recovery feature. 
try this trezor's library out, it is btc/eth blockchain friedly. 
https://github.com/trezor/trezor-firmware/tree/master/crypto