kmackay/micro-ecc #67: Performance issue since multi curve version

juliabory, 2016-02-10:
We use your library in our Cortex M0 hardware. We track the performance on each test build. Unfortunately the graphic shows that the performance has gone down noticeable since commit 0e249d974a02c9783151e932a037b1b6b3450319. Have you planned to work on this? 
We use uECC_OPTIMIZATION_LEVEL=4 is this right or should I change it? Should I add other defines to get better performance? Thanks for your help!
[image]

kmackay, 2016-02-10:
You can try setting uECC_SQUARE_FUNC to 1 as well, it should help a bit.

Probably the biggest improvement would come from fully-inlined multiplication asm (although it would make the binary size quite a bit bigger). I can work on implementing this. Unfortunately I lost my Cortex-M0 test board, so it's a bit hard for me to test at the moment.

juliabory, 2016-02-12:
Thank you so much for your work on microecc. The uECC_SQUARE_FUNC=1 saved us 130ms, now we are only 140ms away from the primary value.