kmackay/micro-ecc #70: ARM Compilation fails with Keil 5.05 and 5.04

ulabs-sgiroux, 2016-02-26:
Hello, I have successfully used your library before.

I decided to update my local copy from what you have published on GitHub. Unfortunately, asm_arm.inc now will not compile under Keil.

I have used preprocessor defines of "uECC_SQUARE_FUNC=1 uECC_OPTIMIZATION_LEVEL=3 uECC_PLATFORM=uECC_arm", but these do not affect the results. The device I am compiling for is a nrf52382 which has a Cortex M4 core.

> compiling uECC.c...
..\..\..\lib\uECC\asm_arm.inc(87): error:  #18: expected a ")"
          : [dptr] REG_RW_LO (result), [lptr] REG_RW_LO (left), [rptr] REG_RW_LO (right),
..\..\..\lib\uECC\asm_arm.inc(95): warning:  #12-D: parsing restarts here after previous syntax error
      );
..\..\..\lib\uECC\asm_arm.inc(52): warning:  #177-D: variable "jump" was declared but never referenced
      uint32_t jump = (uECC_MAX_WORDS - num_words) * 4 * 2 + 1;
..\..\..\lib\uECC\asm_arm.inc(58): warning:  #177-D: variable "left_word" was declared but never referenced
      uint32_t left_word;
..\..\..\lib\uECC\asm_arm.inc(59): warning:  #177-D: variable "right_word" was declared but never referenced
      uint32_t right_word;
..\..\..\lib\uECC\asm_arm.inc(141): error:  #18: expected a ")"
          : [dptr] REG_RW_LO (result), [lptr] REG_RW_LO (left), [rptr] REG_RW_LO (right),
..\..\..\lib\uECC\asm_arm.inc(149): warning:  #12-D: parsing restarts here after previous syntax error
      );
..\..\..\lib\uECC\asm_arm.inc(106): warning:  #177-D: variable "jump" was declared but never referenced
      uint32_t jump = (uECC_MAX_WORDS - num_words) * 4 * 2 + 1;
..\..\..\lib\uECC\asm_arm.inc(112): warning:  #177-D: variable "left_word" was declared but never referenced
      uint32_t left_word;
..\..\..\lib\uECC\asm_arm.inc(113): warning:  #177-D: variable "right_word" was declared but never referenced
      uint32_t right_word;
..\..\..\lib\uECC\asm_arm.inc(1045): error:  #18: expected a ")"
          : [r3] "+l" (num_words), [r4] "=&l" (r4),
..\..\..\lib\uECC\asm_arm.inc(1049): warning:  #12-D: parsing restarts here after previous syntax error
      );
..\..\..\lib\uECC\asm_arm.inc(952): warning:  #177-D: variable "r4" was declared but never referenced
      uint32_t r4, r5, r6, r7;
..\..\..\lib\uECC\asm_arm.inc(952): warning:  #177-D: variable "r5" was declared but never referenced
      uint32_t r4, r5, r6, r7;
..\..\..\lib\uECC\asm_arm.inc(952): warning:  #177-D: variable "r6" was declared but never referenced
      uint32_t r4, r5, r6, r7;
..\..\..\lib\uECC\asm_arm.inc(952): warning:  #177-D: variable "r7" was declared but never referenced
      uint32_t r4, r5, r6, r7;
..\..\..\lib\uECC\uECC.c: 13 warnings, 3 errors

For the time being, I suppose I will use an older version of your code. I am able to and willing to test any tags, branches, or proposed fixes.

kmackay, 2016-02-27:
Looks like your compiler isn't able to parse the constraints syntax. If you can determine what in particular causes the compiler to fail, I can fix it.

ulabs-sgiroux, 2016-02-29:
Thanks for your quick response. I think you are correct -- there is no mention of that in the compiler user manual.

For reference, using the 'small' function results in the same issue, but it is smaller and easier to fit on-screen. I have posted that below 

```
#if !asm_add
uECC_VLI_API uECC_word_t uECC_vli_add(uECC_word_t *result,
                                      const uECC_word_t *left,
                                      const uECC_word_t *right,
                                      wordcount_t num_words) {
    uint32_t carry = 0;
    uint32_t left_word;
    uint32_t right_word;
    
    __asm__ volatile (
        ".syntax unified \n\t"
        "1: \n\t"
        "ldmia %[lptr]!, {%[left]} \n\t"  /* Load left word. */
        "ldmia %[rptr]!, {%[right]} \n\t" /* Load right word. */
        "lsrs %[carry], #1 \n\t"          /* Set up carry flag (carry = 0 after this). */
        "adcs %[left], %[left], %[right] \n\t"   /* Add with carry. */
        "adcs %[carry], %[carry], %[carry] \n\t" /* Store carry bit. */
        "stmia %[dptr]!, {%[left]} \n\t"  /* Store result word. */
        "subs %[ctr], #1 \n\t"            /* Decrement counter. */
        "bne 1b \n\t"                     /* Loop until counter == 0. */
        RESUME_SYNTAX
        : [dptr] REG_RW (result), [lptr] REG_RW (left), [rptr] REG_RW (right),
          [ctr] REG_RW (num_words), [carry] REG_RW (carry),
          [left] REG_WRITE (left_word), [right] REG_WRITE (right_word)
        :
        : "cc", "memory"
    );
    return carry;
}
#define asm_add 1
#endif
```
The issue occurs after RESUME_SYNTAX where it is confused by the ":" notation. This is disappointing and it looks as if there may or may not be an easy fix. FWIW the compiler will let you use variable/parameter names directly http://www.keil.com/support/man/docs/armcc/armcc_chr1359124253408.htm

introiboad, 2016-02-29:
If you're using armcc (which is possible because you're using Keil) then it won't be able to compile it since it doesn't support GCC's asm syntax.
Instead I suggest (I've tried it an it works) compiling uECC.c with GNU ARM Tools Embedded and then linking the .o files to your program in Keil.

carlescufi, 2016-02-29:
ulabs-sgiroux: If you're compiling for the nrf52382 I suggest waiting a few days. A new SDK will be out which comes with an example that supports LE Secure Connections using micro-ecc, with all Makefiles and required project settings included. You will still need to clone the micro-ecc project itself though.

ulabs-sgiroux, 2016-02-29:
Thank you all for your replies. I am targetting both the 52382 and the 51822.

@carlescufi Since you mentioned makefiles, I assume the toolchain will be GCC-based. I may be able to make the switch to GCC tools. I look forward to evaluating LE Secure Connections, but my application may or may not be able to take advantage of that feature.

@introiboad Thank you for the clarification and suggestion. It looks like I may be going open source for this project.

closed 2016-02-29 by ulabs-sgiroux

carlescufi, 2016-02-29:
@ulabs-sgiroux The example project itself will build with Keil, but you will need GCC (which is just a free download from https://launchpad.net/gcc-arm-embedded/+download to build the micro-ecc library.
You will then be able to use micro-ecc for anything you like inside your project and in Keil, with the limitation that you will need attribution to the author for micro-ecc, since you will be distributing a binary that includes it

YoannDqr, 2019-06-25:
Hi ! The only option is always to cros- compile with GCC and links the uECC object file to the Keil project ? Or another option (with Keil) is available now ?

NemoJ, 2019-11-12:
> If you're using armcc (which is possible because you're using Keil) then it won't be able to compile it since it doesn't support GCC's asm syntax.
> Instead I suggest (I've tried it an it works) compiling uECC.c with GNU ARM Tools Embedded and then linking the .o files to your program in Keil.

Hi, introiboad. I am using stm32 with keil and facing the same problem.
Could you please tell me how to compile uECC.c with GNU ARM Tools Embedded and how to use the .o file in my project? 
Looking forward to your reply.

Di

sloanyyc, 2020-04-25:
use this version or pick asn syntax
https://os.mbed.com/users/allankliu/code/uECC/

WilmerSuarez, 2020-12-16:
Hello,

I'm not familiar using GCC. Can someone direct me on how I can compile this and link it to an STM32 Keil project?

Thank you

SALAROX, 2022-06-08:
Hello,
I have the same problem @ulabs-sgiroux , I use Keil 5.36V. Expected a ")" is what I get as error after compilation my project with Micro-ecc library. It relates to asm_arm.inc file. 
Could you please let me know what is the solution?

najmehashemi, 2022-08-14:
i used it it can signature successfully but result of verify is inavalid because r!=rx