kmackay/micro-ecc #73: Tell the byte size of a curve or the max size of all supported curves

mtausig, 2016-03-07:
When I create the array which hold my private/public key, I have to set the array to a more or less elf defined constant value (like 32 in https://github.com/kmackay/micro-ecc/blob/master/test/test_ecdsa.c). It would be much more convenient, if I could  

1. get the value of num_bytes/num_words from uECC_Curve_t
2. get the value of uECC_MAX_WORDS

But I cannot access them as a user, because both use definitions done in uECC.c instead of uECC.h

kmackay, 2016-03-07:
There are the uECC_curve_num_words/bytes/bits() functions in uECC_vli.h. Do those work for you? I could move at least uECC_curve_num_bytes() to uECC.h

mtausig, 2016-03-08:
Ah, ok. I have missed those functions. Yes, they should do the trick if the were moved to uECC.h
Is there a particular reason, why uECC_MAX_WORDS is only defined in uECC.c and not in the header?

kmackay, 2016-03-09:
I don't think exposing uECC_MAX_WORDS is very helpful, since all of the API functions take byte arrays anyway. Presumably in your code you know the largest curve that you support, so you should be able to statically allocate buffers based on that...?

mtausig, 2016-03-09:
Works great, thanks for the patch

closed 2016-03-09 by mtausig