kmackay/micro-ecc #75: Signature not valid with uECC_VLI_NATIVE_LITTLE_ENDIAN

juliabory, 2016-03-14:
I integrated the newest checkin. When I add uECC_VLI_NATIVE_LITTLE_ENDIAN the signature is not valid any more with curve secp256r1. When I remove uECC_VLI_NATIVE_LITTLE_ENDIAN, then the signature is vaild.

carlescufi, 2016-03-14:
The signature will be in little-endian format. When I wrote the patch I ran the signature tests successfully, could it be that the endianness of the signature is the cause here?

carlescufi, 2016-03-14:
I just reran the following test with uECC_VLI_NATIVE_LITTLE_ENDIAN=1 and worked fine:

https://github.com/kmackay/micro-ecc/blob/master/test/test_ecdsa.c

What I did not test is the sign_deterministic() version. Are you using that?

juliabory, 2016-03-14:
I just use the uECC_verify. The signature is received from remote and should be verified on the hardware.

carlescufi, 2016-03-14:
When you enable uECC_VLI_NATIVE_LITTLE_ENDIAN the signature needs to be provided in little-endian format. Are you reversing (or removing the reversal) of the signature bytes you receive from your remote device when enabling this option?

carlescufi, 2016-03-14:
Also remember that the message hash needs to be provided in little-endian as well.

juliabory, 2016-03-14:
it is little endian.

carlescufi, 2016-03-14:
I'm not sure what's going on then. I personally only use key generation and shared secret on my own app, but as mentioned I ran the tests for uECC_sign() and uECC_verify(), which proves that the functions work (at least when executing both on the same platform). All I can think is that the hash used to generate the signature or the signature itself might not be in the same format on both devices.

juliabory, 2016-03-15:
The format should be compatible, because pki is made to be used cross platforms. Should I send you a signature and an matching public key for testing? I hope we can investigate the problem together! Thanks for your work!

carlescufi, 2016-03-15:
Sure, let's do that. I can try it locally and see if we can generate equivalent data

juliabory, 2016-03-15:
Here the data as hex strings:

curve= secp256r1

public key= 18d9002678a869e29748a4968280d18173ad93bc1c56df136828542e524f0ede30da98d729df812fab404c4026ef080685231003b46830b371a37d7e77c0a0e3

signatur= 366e58fcca092f3f81d72676c3a9f864bc7be8fd312951ffc99c6f94c3e100a9032ad867d7c8bc238d591435874c4432af92846854cf3547ab1438c96d2ccce3

-DuECC_ASM=uECC_asm_fast 
-DuECC_PLATFORM=uECC_arm_thumb 
-DuECC_SUPPORTS_secp160r1=0 
-DuECC_SUPPORTS_secp256k1=0 
-DuECC_OPTIMIZATION_LEVEL=4 
-DuECC_ENABLE_VLI_API=1 
-DuECC_SQUARE_FUNC=1 
-DuECC_VLI_NATIVE_LITTLE_ENDIAN

And again: when I skip the last DEFINE it works. Am I doing something wrong?

carlescufi, 2016-03-15:
I think I need the message hash as well, otherwise I have no way to run it myself?

juliabory, 2016-03-15:
Sorry.
hash= 87c8a131120c66c744a0bff41548965ae5d68af7725e8018174a72e55e385bc0

juliabory, 2016-03-15:
The key,signature and hash are the same with and without uECC_VLI_NATIVE_LITTLE_ENDIAN.

juliabory, 2016-03-15:
I think you missed to set the public key? uECC.c lines 1484-1493

carlescufi, 2016-03-15:
This doesn't seem to work with uECC_VLI_NATIVE_LITTLE_ENDIAN=0:

I'm trying to reproduce the problem so that then I can see what's wrong when running with the flag activated.

```
int test_sig(void)
{
    uint8_t pk[] = {0x18,0xd9,0x00,0x26,0x78,0xa8,0x69,0xe2,0x97,0x48,0xa4,0x96,0x82,0x80,0xd1,0x81,0x73,0xad,0x93,0xbc,0x1c,0x56,0xdf,0x13,0x68,0x28,0x54,0x2e,0x52,0x4f,0x0e,0xde,0x30,0xda,0x98,0xd7,0x29,0xdf,0x81,0x2f,0xab,0x40,0x4c,0x40,0x26,0xef,0x08,0x06,0x85,0x23,0x10,0x03,0xb4,0x68,0x30,0xb3,0x71,0xa3,0x7d,0x7e,0x77,0xc0,0xa0,0xe3};
    uint8_t sig[]= {0x36,0x6e,0x58,0xfc,0xca,0x09,0x2f,0x3f,0x81,0xd7,0x26,0x76,0xc3,0xa9,0xf8,0x64,0xbc,0x7b,0xe8,0xfd,0x31,0x29,0x51,0xff,0xc9,0x9c,0x6f,0x94,0xc3,0xe1,0x00,0xa9,0x03,0x2a,0xd8,0x67,0xd7,0xc8,0xbc,0x23,0x8d,0x59,0x14,0x35,0x87,0x4c,0x44,0x32,0xaf,0x92,0x84,0x68,0x54,0xcf,0x35,0x47,0xab,0x14,0x38,0xc9,0x6d,0x2c,0xcc,0xe3};
    uint8_t hash[] = {0x87,0xc8,0xa1,0x31,0x12,0x0c,0x66,0xc7,0x44,0xa0,0xbf,0xf4,0x15,0x48,0x96,0x5a,0xe5,0xd6,0x8a,0xf7,0x72,0x5e,0x80,0x18,0x17,0x4a,0x72,0xe5,0x5e,0x38,0x5b,0xc0};
    const struct uECC_Curve_t *curve = uECC_secp256r1();
    int ret;

    ret = uECC_verify(pk, hash, sizeof(hash), sig, curve);

    return ret;
}
```

carlescufi, 2016-03-15:
> I think you missed to set the public key? uECC.c lines 1484-1493

No, it's set in line 1472

carlescufi, 2016-03-15:
OK now with the new public/signature/hash that you have uploaded it works with the flag disabled

carlescufi, 2016-03-15:
OK this code below now works fine with and without uECC_VLI_NATIVE_LITTLE_ENDIAN:

```
static void reverse(uint8_t* p_arr, uint32_t len)
{
    uint32_t i;

    for (i = 0; i < len/2 ; i++)
    {
        uint8_t tmp = p_arr[i];
        p_arr[i] = p_arr[len - i - 1];
        p_arr[len - i - 1] = tmp;
    }
}

int test_sig(void)
{
    uint8_t pk[64]   = {0x18,0xd9,0x00,0x26,0x78,0xa8,0x69,0xe2,0x97,0x48,0xa4,0x96,0x82,0x80,0xd1,0x81,0x73,0xad,0x93,0xbc,0x1c,0x56,0xdf,0x13,0x68,0x28,0x54,0x2e,0x52,0x4f,0x0e,0xde,\
                        0x30,0xda,0x98,0xd7,0x29,0xdf,0x81,0x2f,0xab,0x40,0x4c,0x40,0x26,0xef,0x08,0x06,0x85,0x23,0x10,0x03,0xb4,0x68,0x30,0xb3,0x71,0xa3,0x7d,0x7e,0x77,0xc0,0xa0,0xe3};
    uint8_t sig[64]  = {0xef,0x15,0x07,0xd5,0x4e,0x7f,0xef,0x8d,0x11,0x7d,0x25,0x41,0x81,0x1c,0x5e,0x3f,0x33,0xec,0x85,0x18,0xa0,0x89,0x8a,0x21,0x2b,0x63,0x9a,0xb6,0x2c,0xd5,0x7d,0x2d,\
                        0xe3,0xca,0x4a,0x9c,0x08,0x1f,0x7b,0x91,0x1c,0x8a,0xd5,0xed,0x84,0xae,0x26,0xb8,0xd9,0xdd,0x86,0x2d,0xa0,0x08,0x12,0x5c,0x5c,0xfe,0x45,0x64,0x33,0x8c,0x44,0xae};
    uint8_t hash[32] = {0x09,0xdb,0xf8,0x35,0x7c,0x27,0x9b,0x43,0x92,0x06,0x46,0xb8,0xb1,0xa8,0x74,0x55,0x01,0x43,0x3f,0x5e,0xf6,0x1e,0x08,0x01,0x6f,0x4f,0x34,0xca,0x01,0xc8,0x8e,0x84};
    const struct uECC_Curve_t *curve = uECC_secp256r1();
    int ret;

#if uECC_VLI_NATIVE_LITTLE_ENDIAN
    reverse(pk, 32);
    reverse(pk + 32, 32);

    reverse(sig, 32);
    reverse(sig + 32, 32);

    reverse(hash, 32);
#endif
    ret = uECC_verify(pk, hash, sizeof(hash), sig, curve);

    return ret;
}
```

carlescufi, 2016-03-16:
@juliabory Did you try the snippet above out? As I suspected the keys you provided are in big endian, and converting them to little endian works. If that fixes it for you please close the issue.

juliabory, 2016-03-16:
Yes this works! Thanks!

closed 2016-03-16 by juliabory