kmackay/micro-ecc #76: How to change ECC curve Parameter

hungnguyen88, 2016-03-15:
Hi,

May I change the curve Parameter a in the file curve-specific.inc?
For example in the file curve-specific.inc, I see function uECC_Curve_t curve_secp160r1()
I only see I can change the parameter p, b, G, n
About G point, whether the first chain is Gx and the second chain is Gy.
Thank you a lot,

Hung Nguyen.

kmackay, 2016-03-15:
The code assumes that a is -3 (except for secp256k1, where a is 0). If you want a different a, you will need to implement double_jacobian() and x_side() for your curve.

For G, yes, it is X, then Y.

closed 2016-03-15 by kmackay

hungnguyen88, 2016-03-16:
Thank you,
I find I can change _3[uECC_MAX_WORDS] = {3} to my a parameter in x_side_default() function.
However I don't understand: How about double_jacobian(), what should I change in this function. The A parameter I want to change is "a = 340E7BE2A280EB74E2BE61BADA745D97E8F7C300" 

Because I want to change Curve Parameter like that example:
p = E95E4A5F737059DC60DFC7AD95B3D8139515620F
a = 340E7BE2A280EB74E2BE61BADA745D97E8F7C300
b = 1E589A8595423412134FAA2DBDEC95C8D8675E58
x = BED5AF16EA3F6A4F62938C4631EB5AF7BDBCDBC3
y = 1667CB477A1A8EC338F94741669C976316DA6321
n = E95E4A5F737059DC60DF5991D45029409E60FC0

Thank your help.