kmackay/micro-ecc #77: openssl interoperability

henomis, 2016-03-15:
I'm trying to have an interoperability between micro-ecc and openssl. 

What I do is create a key pair with openssl

$ openssl ecparam -genkey -name secp256r1 -out k.pem

view the key pair:

$ openssl ec -in k.pem -noout -text

Private-Key: (256 bit)
priv:
    00:de:e6:58:5e:d3:54:13:0c:9f:e3:91:cc:c5:89:
    6d:78:75:88:91:37:ef:7d:e4:31:f9:de:93:97:2a:
    09:da:8a
pub: 
    04:b8:11:4b:94:73:4e:2b:f9:0b:77:22:7f:c0:bd:
    d7:91:bc:7b:e5:70:1a:d1:9c:16:51:de:fc:a0:87:
    40:bc:55:d2:cc:0b:57:6b:1f:3d:1e:fa:c9:f7:a5:
    38:b5:96:46:33:ce:c0:cf:39:42:0f:b6:a1:78:fc:
    ec:06:f9:26:9d
ASN1 OID: prime256v1
NIST CURVE: P-256

Here is the first problem: pub key is 65 bytes long and priv 33 bytes. They should be 32 and 64, isn't? 
How can I create a key pair in openssl and import into micro-ecc?

juliabory, 2016-03-15:
You have to cut the first byte I think and then you have your x,y point.

closed 2016-04-21 by kmackay

stefanos1019, 2016-07-17:
@henomis did you get this to work?

michalfapso, 2017-06-02:
Thank you @juliabory, when I removed the first byte from both private and public keys, micro-ecc started to work correctly with my openssl keys.

Note: I used uECC_valid_public_key() and uECC_compute_public_key() functions for checking the keys.