kmackay/micro-ecc #79: LPC support?

lisa2code, 2016-04-04:
In the official website: http://kmackay.ca/micro-ecc/
It states that it is "Reasonably fast: on an LPC1114".

But the software does not build as is with the official LPCXpresso IDE for this platform:

http://www.nxp.com/products/software-and-tools/hardware-development-tools/lpcxpresso-boards/lpcxpresso-ide:LPCXPRESSO

Is there any working LPCXpresso IDE project to build it?

kmackay, 2016-04-05:
I just built using gcc; I've never used LPCXpresso.

kmackay, 2016-04-06:
It seems to work fine for me in LPCXpresso. Here are the steps I followed:
 1. Create a new project, I made a "C Project".
 2. Import the micro-ecc code into the project. I just imported uECC.c as all .h and .inc files.
 3. Build. In the default uECC configuration, you need to build in Release mode, otherwise it will complain about impossible asm constraints. If you want to use Debug mode, you can comment out the inline assembly for uECC_vli_mult() in asm_arm.inc.

closed 2016-04-06 by kmackay