kmackay/micro-ecc #81: EDHC Co-factor multiplication

Seechay, 2016-05-31:
I was wondering if you could help me provide support for co-factor multiplication for shared keys. I'm trying to add ECIES to this, but the standard specifies co-factor. I can't tell if it's already being done, so if it is please close this. What I've done in the mean time is add the following to uECC_shared_secret()
	`EccPoint_mult(public, public, cofactor, 0, curve->num_n_bits + 1, curve);`
Cofactor is defined `uECC_word_t cofactor[1]; cofactor[0] = 1; ` But when I generate the secret key, it's always different. Commenting out the initial_Z code from the rng, results in a static key. 

On another note, what does this section of code do? 
```
/* If an RNG function was specified, try to get a random initial Z value to improve
       protection against side-channel attacks. */
    if (g_rng_function) {
        if (!uECC_generate_random_int(p2[carry], curve->p, num_words)) {
            return 0;
        }
        initial_Z = p2[carry];
		printf("\ninitial_Z: %d",initial_Z);
    }
```
Commenting it out has no effect on the secret key if my code change is ommitted.

kmackay, 2016-06-01:
Generating a random Z-value provides some protection against side-channel attacks, but doesn't affect the final result.

Seechay, 2016-06-01:
I thought that was the case, it was modifying the result when I tried to multiply the point again by 1, for the co-facator. Do you have any suggestions for the co-factor, or am I doing it the correct way?

kmackay, 2016-06-02:
I'm not sure exactly what you're trying to do. You shouldn't need to multiply by 1 (since x * 1 = x).

Going by the Wikipedia definitions for ECIES, you can:
 * Generate r and R using uECC_make_key()
 * Generate S using uECC_shared_secret()

kmackay, 2016-06-02:
ie, you shouldn't need to modify the code at all to implement ECIES

Seechay, 2016-06-02:
I'm looking to implement a standard, which specifies the use of ECDHC for shared keys (which is the same as ECDH but with the cofactor multiplication by 1) instead of ECDH. Unless I'm not doing the x * 1 correctly, I'm getting a different result instead of the input. I'm not completely aware of the properties of elliptic curve math, but if 1 * x = x,  then I don't need to do it and thus issue can be closed.

Seechay, 2016-06-02:
Closing the issue. Misinterpreting the api, I thought `EccPoint_mult` was used to multiply two vli. Using `uECC_vli_modMult_fast` fixes my issue.

closed 2016-06-02 by Seechay

kmackay, 2016-06-03:
BTW, when the cofactor is 1, there is no difference between ECDH and ECDHC.

Seechay, 2016-06-03:
I realized that today. I'm still trying to grasp the elliptic curve math,
it's pretty different. Luckily all the curves I need to use have a cofactor
of 1!