kmackay/micro-ecc #84: Use HMAC

mathDv, 2016-07-20:
Hi. First, congrats for your work.
I'm looking for a implementation in C of HMAC. It is possible to use in micro ecc ?
Thanks in advance.

kmackay, 2016-07-22:
micro-ecc has an implementation of HMAC which you are free to use. You need to supply the hash function though (eg SHA256). One implementation of SHA256 can be found here: http://sources.freebsd.org/RELENG_8/src/sys/crypto/sha2/sha2.c <http://sources.freebsd.org/RELENG_8/src/sys/crypto/sha2/sha2.c>

See my test/test_ecdsa_deterministic.c.example to see how I wired up the SHA256 code to the micro-ecc HMAC. The actual HMAC implementation is in uECC.c; see HMAC_init(), HMAC_update(), and HMAC_finish().

mathDv, 2016-07-23:
Hi. 
I could make the code run with this hash implementation (https://github.com/B-Con/crypto-algorithms).
Very nice, Thanks for the help..

closed 2016-07-23 by kmackay