kmackay/micro-ecc #87: uECC_verify not working with persisted private/public keys

JayBeavers, 2016-08-28:
I've written a partial wrapper around micro-ecc for NodeJS to enable me to use your library rather than the more involved crypo API which is based on OpenSSL.

Each time my app starts, my wrapper call:

const struct uECC_Curve_t* curve = uECC_secp160r1();

and then I call a combination of:

    uint8_t result = uECC_make_key(publicKeyBufferData, privateKeyBufferData, curve);
    uECC_sign(privateKeyBufferData, inputBufferData, inputBufferLength, signatureBufferData, curve);
    uint8_t result = uECC_verify(publicKeyBufferData, inputBufferData, inputBufferLength, signatureBufferData, curve);

Everything works as expected for the duration of the process.  Then I added file persistence to privateKeyBufferData/publicKeyBufferData so I was loading pregenerated keys upon restart rather than calling uECC_make_key.  Once I started doing this uECC_sign/uECC_verify stopped working.

I've validated that the private/public key data is being correctly persisted/restored.  Is there some additonal state, perhaps generated by uECC_secp160r1() that needs to be persisted along with the keys?

Repro environment is OSX using the compiler settings generated by gyp, modified for C++14.  uECC is being wrapped by a C++ library that bridges the C into the Google V8 environment.

JayBeavers, 2016-08-28:
Ah, I didn't check your functions uECC_curve_private_key_size which was 21 bytes, not 20 bytes.  Fortunately I went browsing through your closed bugs and found the conversation on that.  Once I changed private key to 21 bytes, all started working as expected.

Thanks for the great library!

closed 2016-08-28 by JayBeavers

kmackay, 2016-08-29:
Thanks for commenting that you fixed the issue.