kmackay/micro-ecc #88: Watchdog hook for MCU

weiqj, 2016-09-28:
I am running uECC on a cortex M3 MCU. Currently I don't enable watchdog, with maximum timeout of 1 second.
However I will enable watchdog soon and I am afraid that the long processing time of uECC may possible cause watchdog timeout.
I am wondering if it is possible to add a watchdog hook in the loop code of uECC. Of course I can always modify the code by myself but I think it is better to make it official.

jnysen, 2016-09-28:
Each operation takes about 400ms on a 16MHz Cortex M0 which only has a 16x16 (effective) multiplier (it's actually 32x32, but only keeps the least significant word) . The M3 should be much faster with a proper 32x32 multiplier, which means you can clear the watchdog after each uECC operation and still be well within your maximum 1 second timeout.

That being said, the best way to deal with long/complex operations (and watchdogs in general) is to setup a watchdog Timer interrupt (or thread if you're using an RTOS) that is the sole entity in charge of the hardware watchdog. In the handler for that interrupt (or thread), you can periodically check various things like stack sentinels, guard variables, state variables, and have separate down-counters for various long operations (some of which may even be interrupt driven (or in a separate thread)).

This also lets you log a more detailed failure mode description before resetting (assuming the interrupt itself isn't what has failed) for improved fault detection in a deployed system. Which is especially useful when you are seeing sporadic infrequent resets in the device.

The key here is to only ever use the hardware watchdog reset instruction in one place in the handler/thread. Everywhere else you just set one of the soft 'watchdog' countdown variables back to the desired 'timeout' value. You can even let the zero value disable a down-counter, and only force a reset if one of active counters transitions from 1 to 0.

It is usually not a good idea to reset a watchdog (or even a soft timeout) in the middle of a complex loop operation just to keep the watchdog happy. Doing that reduces the ability to capture random or data dependent issues that may cause a complex loop to become infinite or excessively long. (eg. A control/test variable getting into a pathological infinite looping state due to memory corruption. Or bad data causing an outer loop to execute 4+ billion times instead an expected much smaller amount, causing the loop to take days/months/years to return instead of milliseconds.)

kmackay, 2016-11-06:
I don't plan to add any hooks to the point multiplication loop.

closed 2016-11-06 by kmackay