kmackay/micro-ecc #89: Sign with openSSL & verify with micro-ecc

satishlvr, 2016-10-31:
Hi,

My scenario is generate key pair using openSSL, sign with openSSL & do verification with micro-ecc.
Can you please share if you have any sample code for this ?
My confusion was how we can convert key structure of openSSL to suit as input to "uECC_verify" in micro-ecc. As per my understanding, we should not pass any encoded buffer directly to this API. so, I am confused how to extract the public key from openSSL structure and save and then pass to application which verifies using micro-ecc.

Request your help regarding this.

Thank you !

Regards
Satish

kmackay, 2016-11-06:
I don't recall exactly which OpenSSL function you use to extract the public key; but I know it is possible. uECC_verify() takes the standard form of uncompressed public key (although, you should remove any 0x04 prefix byte; the public key should be exactly 2 time the curve size, eg 64 bytes for a 256-bit curve).

closed 2016-11-06 by kmackay

robwbowman, 2017-05-01:
Hi,

Same scenario, generated key pair using OpenSSL, signed with OpenSSL and perform verification with micro-ecc. 

In OpenSSL - create private/public keys, sign and verify:

openssl ecparam -name secp256k1 -genkey -out private.pem
openssl ec -in private.pem -pubout -out public.pem
echo|set /p="ABCDEFGHIJKLMNOPQRSTUVWXYZ012345" > message.dat
openssl dgst -ecdsa-with-SHA1 -sign private.pem -out signature.dat message.dat
openssl dgst -ecdsa-with-SHA1 -verify public.pem -signature signature.dat message.dat

In Python/ecdsa - read OpenSSL public-key and verify signature:

from ecdsa import VerifyingKey, util, SECP256k1
from hashlib import sha1

vk = VerifyingKey.from_pem(open("public.pem").read())
message = open("message.dat","r").read()
signature = open("signature.dat","rb").read()
vk.verify(signature, message.encode(), hashfunc=sha1, sigdecode=util.sigdecode_der)

Using Python the OpenSSL private/public, hash and signature were written to micro-ecc C-Language arrays and uECC_verify fails. 

An observation, when signing with 32-bytes being "ABCDEFGHIJKLMNOPQRSTUVWXYZ012345" a 71-byte signature is created. The example test_ecdsa.c wants a 64-byte signature.

Any ideas?

With regards,

Rob Bowman

jbdatko, 2017-05-01:
On Mon, 2017-05-01 at 07:15 -0700, robwbowman wrote:
> 
> sigdecode=util.sigdecode_der)

^^ Are you accounting for the extra bytes that ASN.1 encoding will add
to the signature. uECC works on the raw byte-vector whereas most other
software works in ASN.1. 

I think they add: <tag (octect string 0x04 or 0x30, can't remember)>
<length> <value> Where the value is then further proceeded by the
uncompressed point tag <0x04>. After you strip all that way though, you
should find the raw R,S of the signature.

Anyway, this is just a driveby. Hope it helps.

robwbowman, 2017-05-05:
Hi Josh,

I will thanks for the insight and direction. I will look into how the
signature is encoded and output a raw byte-vector.

Regards,

Rob

michalfapso, 2017-06-02:
I had the same problem with the ASN.1 signatures. I just copied the ASN.1 signature generated by openssl to this online converter: https://holtstrom.com/michael/tools/asn1decoder.php
Here is its output:
```
SEQUENCE {
   INTEGER 0x50195fd2984c37f6f0ccb2cd4363418dbba013b6139aed45c9148198cb7c9c88
   INTEGER 0x2da66401ee9b6aca65698da950e7bcad6817f9b03175f63e07a5fbd261426cd9
}
```
so the raw signature compatible with micro-ecc is:
```
50195fd2984c37f6f0ccb2cd4363418dbba013b6139aed45c9148198cb7c9c882da66401ee9b6aca65698da950e7bcad6817f9b03175f63e07a5fbd261426cd9
```