kmackay/micro-ecc #91: Java Interop

DoDoENT, 2016-12-13:
Hi!

I am using this library in my C++ project on embedded device where some data needs to be signed. For hashing, I am using SHA3 implementation from [Stephan Brumme](http://create.stephan-brumme.com/hash-library/). The signature that is generated from ECC library can be correctly verified by the same library with `uECC_verify`, however my use case requires me to verify the signature on server which uses Java.

Unfortunately, Java makes it rather complicated to import public keys and signatures as raw byte array due to endianess. Therefore, I would like to know what is the endianness of both public key and signature byte arrays that are returned from your library. Java requires big endian arrays (I think) and I assume that your library returns them in little endian (especially if compiled with `-DuECC_VLI_NATIVE_LITTLE_ENDIAN=1`. However, I am confused about endianness when library is not compiled with that option (default is 0). I tried treating the result public key as both little and big endian arrays and for some keys in both interpretations I get exception when loading the public key. The exception says that _x value is invalid_, although some decompilation of Java code and debugging has lead me to information that actually y value is invalid in public key (Bouncy castle requires that both x and y values are positive and less than Q, which is the value `p` in `uECC_Curve_t struct`) - in my case the y value of my public key is negative.

I've attached a small [java example project](https://github.com/kmackay/micro-ecc/files/648378/ECDSATest.zip) which demonstrates the issue I am having. The public key in question is `50021df8eaab8fdbd1d3280ab6877c9526c3bf28ee55716dab3129d981af4e7317ee2063c3f16d521cc2c752ffe45b493271a89c92a62cf0ab794a0b365a19eb`, basically `x=50021df8eaab8fdbd1d3280ab6877c9526c3bf28ee55716dab3129d981af4e73` and `y=17ee2063c3f16d521cc2c752ffe45b493271a89c92a62cf0ab794a0b365a19eb`. What I don't know is whether the most significant byte of Y is `17` or `eb` (using `eb` makes Y negative, which does not work for java, using `17` makes signature verification failed.

Does anyone have any experience with checking microECC-generated signatures in Java (either with Bouncy Castle or some other library)? If I do not make it work with BC, I am really leaning towards implementing `uECC_verify` directly in Java and use it on my server or use JNI on server to call `uECC_verify` directly.

DoDoENT, 2016-12-13:
OK, in the meantime I found my way around Bouncy Castle and managed to correctly import compressed public key and signature from ECCs output. For future reference, I am pasting the code here:

```java
	public static byte[] toByte(String hex) {
		return Hex.decode( hex );
    }
	
	private static ECPublicKeyParameters getPublicKeyParameters(byte[] pubKey, String curveName) throws NoSuchAlgorithmException, InvalidKeySpecException, NoSuchProviderException, InvalidParameterSpecException {	
		X9ECParameters params = SECNamedCurves.getByName( curveName );
		ECDomainParameters domainParams = new ECDomainParameters( params.getCurve(), params.getG(), params.getN() );

		System.out.println( "Curve N:       " + params.getN() );
        ECPublicKeyParameters ecPublicParams = new ECPublicKeyParameters( params.getCurve().decodePoint(pubKey), domainParams );
        return ecPublicParams;
	}
	
	private static boolean checkSignature( String message, String providedSignature, String publicKey ) throws UnsupportedEncodingException, NoSuchAlgorithmException, InvalidKeySpecException, NoSuchProviderException, InvalidParameterSpecException {
		ECDSASigner signer = new ECDSASigner();
		
		byte[] pubKey = toByte( publicKey );	
		
		byte[] invertedPubKey = new byte[ pubKey.length ];
		for( int i = 0; i < pubKey.length; ++i ) {
			invertedPubKey[ i ] = pubKey[ pubKey.length - 1 - i ];
		}
		
		ECPublicKeyParameters ecPublicKeyParameters = getPublicKeyParameters( pubKey, "secp256k1" );
		ECPoint Q = ecPublicKeyParameters.getQ();
		
		System.out.println( "Public key X = " + Q.getXCoord().toBigInteger() );
		System.out.println( "Public key Y = " + Q.getYCoord().toBigInteger() );
		
		signer.init( false, ecPublicKeyParameters );
		
		byte[] rawBytes = toByte( providedSignature );
		byte[] r = new byte[ rawBytes.length / 2 ];
		byte[] s = new byte[ rawBytes.length / 2 ];
		System.out.println( "Signature raw bytes length is " + rawBytes.length );
		for( int i=0; i < rawBytes.length / 2; ++i ) {
			r[ i ] = rawBytes[ i ];
			s[ i ] = rawBytes[ rawBytes.length / 2 + i ];
		}
		BigInteger br = new BigInteger( 1, r );
		BigInteger bs = new BigInteger( 1, s );
		
		System.out.println( "Signature R: " + br );
		System.out.println( "Signature S: " + bs );
		
		// calculate SHA-3 hash of message
		SHA3.DigestSHA3 hasher = new SHA3.DigestSHA3(256);
		hasher.reset();
		hasher.update(message.getBytes());
		byte[] hash = hasher.digest();
		
		String hexHash = Hex.toHexString(hash);
		System.out.print( "SHA3-256 hash of '" + message + "' is: " );
		System.out.println( hexHash );
		
		return signer.verifySignature( hash, br, bs );
	}
```
Sorry for being a noob.

closed 2016-12-13 by DoDoENT

trupheenix, 2017-12-06:
Why did you even invert the public key? I see you haven't used it at all.

DoDoENT, 2017-12-06:
> Why did you even invert the public key? I see you haven't used it at all.

That has probably left from my previous attempts when I tried to figure out what is wrong - I'll clean the java code.

FYI, the problem was not in endianess - it was in a fact that java does not support unsigned types, so as soon as first bit of first byte of public key was 1, Java imported that as negative number, which is of course invalid for ECC.

trupheenix, 2017-12-07:
Currently I am struggling with the reverse. ie. Sending public key and signature from Java. Would you have any pointers on how the data needs to be sent? 

I just filed this day before yesterday. 

https://github.com/kmackay/micro-ecc/issues/130

DoDoENT, 2017-12-09:
> Currently I am struggling with the reverse. ie. Sending public key and signature from Java. Would you have any pointers on how the data needs to be sent?

Unfortunately, I don't have experience with this. As far as I know, signature is consisted of two numbers R and S, so you can export them from java as byte array consisting of two big integers (reverse of what I've done above). The public key is similarly represented with two numbers: X and Y and can be compressed in a way that it only contains X and exponent required to calculate Y from X. Normally, the full public key should be exported in a same way as signature. Whether or not there are caveats with doing that, I wouldn't know.