kmackay/micro-ecc #92: key generation on different platforms

dalbert2, 2017-02-02:
First, thank you for the awesome library!

I built a utility application under cygwin64 and the application works well; it can generate keys, validate them, sign and verify data, etc.

I've also built an embedded application for Cortex M0+ and unfortunately, public keys generated by the cygwin64 application are seen as invalid by the Cortex application.  

Both apps are using secp256k1.

For example uECC_valid_public_key() considers the following public key valid on the x86_64 platform but invalid on the Cortex M0+ platform.

Public key: FB 32 76 85 5D 1A B5 04 18 7C B4 26 4A F9 20 5A
            E3 43 E3 BF D3 5C FB 29 D7 82 A9 33 43 F9 3F 68
            1E 02 28 78 12 95 A5 A1 4F 6D 45 15 A8 0E 39 FC
            BA E8 F6 A7 1B B7 5E 60 E4 72 94 2A 3E 2B 1C 4E

The Cortex build defines the following:
-    uECC_PLATFORM=uECC_arm_thumb
-    uECC_WORD_SIZE=4
-    uECC_OPTIMIZATION_LEVEL=2
-    uECC_VLI_NATIVE_LITTLE_ENDIAN=1
-    uECC_SQUARE_FUNC=1

Any suggestions would be appreciated.

dwalkes, 2017-02-02:
Hi @dalbert2, I noticed the same thing about endianness flag being related to the endian requirements for the public key.  I think there's a mention of this [here](https://github.com/kmackay/micro-ecc/blob/14222e062d77f45321676e813d9525f32a88e8fa/uECC.h#L40)
 `If enabled (defined as nonzero), this will switch to native
little-endian format for *all* arrays passed in and out of the public API. This includes public 
and private keys, shared secrets, signatures and message hashes. `
Not stated but implied is that the keys, signatures, and hashes should be big endian format if uECC_VLI_NATIVE_LITTLE_ENDIAN is 0.
I've got a fork of the nordic semiconductor tool at https://github.com/Trellis-Logic/pc-nrfutil which can output keys in either format if you are interested.  See instructions at https://github.com/Trellis-Logic/sdu/blob/master/platforms/redbear_duo/platform.md 
`nrfutil keys generate myprivatekey.pem
nrfutil keys display --key pk --format code myprivatekey.pem`
This will spit out keys suitable for include in your c file in either endian format which can be used for either setting of uECC_VLI_NATIVE_LITTLE_ENDIAN.

dalbert2, 2017-02-02:
Thanks Dan, I wrote a similar utility in cygwin to generate/test/sign/verify/etc.   I would expect keys to be platform independent and had assumed the #define for endianess was only for code optimization as the comment above it implies.  If the keys are really endian-specific, that's a serious issue if uECC is used to encrypt communications since it means the receiver needs to know how the key was generated; in the case of code signing it's probably OK since the code is always for some target processor, but if keys are really endian-specific, that should be front-and-center in the documentation and comments.  

Also, although much less important, given the prevalence of little-endian machines, that should probably be the default.

carlescufi, 2017-02-02:
uECC_VLI_NATIVE_LITTLE_ENDIAN affects input keys and output results. The standard format for security keys is big endian, but certain protocols such as Bluetooth Low Energy use little endian over the air. I added this macro to avoid having to convert between the protocol format and the input format for micro-ecc.

dalbert2, 2017-02-02:
Thank you Carles.  I am used to network byte order for everything; it's interesting that BLE changed to little endian over the air.  I now understand the issue and uECC is working beautifully; I think it would be good to update the comments for NATIVE_LITTLE_ENDIAN to make it clearer that keys generated using LE and BE are incompatible with each other so all parties must use the same endian-ness.  I don't know if the code is open to editing by anyway, if it is, I'll update the comment...otherwise, I would encourage someone who has appropriate permissions to add this to save others the problem.

kmackay, 2017-05-21:
I added a note in uECC.h

closed 2017-05-21 by kmackay