kmackay/micro-ecc #93: Patents?

dalbert2, 2017-02-06:
Has anyone looked into whether using micro-ECC would cause an application to infringe on the [Certicom patents](https://en.wikipedia.org/wiki/ECC_patents)?

jbdatko, 2017-02-06:
IANAL and probably you need a lawyer to give you counsel re: anything with patents, but you might find this useful: https://www.rfc-editor.org/rfc/rfc6090.txt

kmackay, 2017-05-21:
You should check with a lawyer; I can't make any guarantees about patent infringement.

kmackay, 2017-05-21:
I don't know of any unexpired patents covering the micro-ecc code though.

closed 2017-05-21 by kmackay

dalbert2, 2017-05-22:
Thank you...and thank you again for micro-ecc!

weiqj, 2017-05-22:
What about point compression? I still avoid it in my products because I
have concerns.
Appreciate your effort very much.

Best
Qingjun

or mute
the thread

msembinelli, 2017-06-02:
As per weiqj's comment, I am also curious. Here is the patent in question:
https://www.google.com/patents/US6252960

It expires in August 2018