kmackay/micro-ecc #96: 16-bit Implementation (MSP430FR5994)

mjasonVT, 2017-03-07:
Thank you for your work on micro-ECC.  I was reading in the past issues (#23) that you could possibly make a 16-bit version for the MSP430.  Is this still a possibility?

Thanks!

Jason

cladmi, 2017-03-29:
Hello, I have the same concern.

The project [eclipse tinydtls](https://projects.eclipse.org/projects/iot.tinydtls) plans to use micro-ecc instead of its own ecc implementation. However, switching to micro-ecc would currently drop support to 16bit architectures. If micro-ecc would have a 16bit support, even if it was a really slow version at first, would help using it only without requiring maintaining two versions.

Would you merge a 16bit support ? if yes, what do you think of the current state of #23 repository ? What additional work would you require to be done on it before merging.

Thank you in advance.

Regards,
Gaëtan

kmackay, 2017-04-02:
I don't mind merging 16-bit support. It would be very slow though without
any asm optimizations.