kmackay/micro-ecc #97: micro ecc create wrong publickey in windows

EricYangCn, 2017-03-10:
I created a pair of key in nrf52832(arm), and checked it in android,however failed in checking in windows. I noticed that though I do select windows, I didn't get the correct result, why? My vs2013(ce) code attached:
// mircoecctest.cpp : Defines the entry point for the console application.
#include "stdafx.h"
#include "uECC.h"

int _tmain(int argc, _TCHAR* argv[])
{
	uint8_t privatekey[] = { 
		0x76, 0xA1, 0x3D, 0x5A, 0x97, 0xA4, 0x09, 0x2C,
		0x29, 0xCF, 0x0F, 0xDA, 0xA7, 0x3C, 0x6F, 0xED,
		0x92, 0x93, 0x62, 0xEE, 0x21, 0xF8, 0xA5, 0x40,
		0x53, 0x33, 0x19, 0xB1, 0x82, 0xAE, 0xCE, 0x1C};
	 uint8_t  publickey[] = { //The result in nrf52832(10040)
		0x80, 0x73, 0xC2, 0xA3, 0xB6, 0xC7, 0xF8, 0x61,
		0xBC, 0x0D, 0xCB, 0x68, 0x3C, 0x83, 0xBD, 0xCE,
		0x48, 0xB8, 0x07, 0x1C, 0x7E, 0xCC, 0x39, 0x38,
		0x7C, 0x9C, 0xC7, 0xBF, 0x91, 0xC0, 0xCE, 0x25,
		0xF1, 0xD2, 0xC2, 0x0F, 0x30, 0x4B, 0x98, 0xC4,
		0x74, 0xCF, 0xDB, 0x01, 0x5C, 0xC6, 0xD4, 0x6D,
		0x19, 0x90, 0x30, 0x0A, 0x40, 0xA0, 0xBD, 0xC2,
		0x2E, 0x9C, 0xAA, 0x8E, 0x46, 0xC7, 0x92, 0x3D};
	 uint8_t publickeywin32[64];
#if (defined(_WIN32) || defined(_WIN64))
	 printf("windows is selected\r\n");
#endif
	 uECC_compute_public_key(privatekey, publickeywin32, uECC_secp256r1());
	printf("public key is :\r\n");

	for (int i = 0; i <64; i++)
	{
		printf(" %02X", publickeywin32[i]);
		if (i % 16 == 15)
			printf("\r\n");
	}
	getchar();
	return 0;
}
The output data I got:

windows is selected

public key is :

 7A 93 1E D8 1B 4A 32 29 88 7A A5 3C A2 F9 BA 3F

 B4 83 2A DB 0B EF E3 F3 2A E4 FA 9D B1 38 00 99

 A4 4F 7D 1D F4 98 90 E4 9B 10 0C 4D D8 95 45 A9

 90 34 F8 95 A3 A6 31 79 D2 BD 1A 19 FF E1 BB E0

EricYangCn, 2017-03-25:
I found my miatake already,  my solution is:

There is a define statement in uEcc.h(line48-line50) :
ifndef uECC_VLI_NATIVE_LITTLE_ENDIAN
#define uECC_VLI_NATIVE_LITTLE_ENDIAN 0
endif

When we compile codes in Keil5, it's right. However, in windows, it doesn't work. If we change the value of uECC_VLI_NATIVE_LITTLE_ENDIAN from 0 to 1, it works well ,at least in VS2013. So what we should do is :

ifndef uECC_VLI_NATIVE_LITTLE_ENDIAN
#define uECC_VLI_NATIVE_LITTLE_ENDIAN 1
endif

dwalkes, 2017-03-25:
Hi Eric,
I was also initially confused about this.   The issue is that the endian expectation changes for keys when the value of this define changes.  See the comment [in uECC.h](https://github.com/kmackay/micro-ecc/blob/master/uECC.h#L40)

> If enabled (defined as nonzero), this will switch to native
little-endian format for *all* arrays passed in and out of the public API. This includes public 
and private keys, shared secrets, signatures and message hashes

If you want to support both #define settings for uECC_VLI_NATIVE_LITTLE_ENDIAN in your build you should also conditionally define your key or swap endian between big and little as needed based on the build switch.

In case it's useful for you, I've made some modifications to Nordic's pc-nrfutil in [my branch](https://github.com/Trellis-Logic/pc-nrfutil) which creates keys in the correct endian format for either setting using a #ifdef uECC_VLI_NATIVE_LITTLE_ENDIAN build switch using command

> nrfutil keys generate private.pem
> nrfutil keys display --key pk --format code private.pem

closed 2017-05-21 by kmackay