kmackay/micro-ecc #103: Why curve_secp160r1.num_n_bits = 161?!

edogaldo, 2017-05-09:
Hello @kmackay, maybe a silly question, I'm not in with elliptic courves but I was noticing that only for curve_secp160r1 num_n_bits is different from what specified in the name.
Is it correct?

Thanks in advance and greeting for this very interensting library!

E.

edogaldo, 2017-05-09:
Other question: where is "a" in the curves parameters?

kmackay, 2017-05-21:
The value of n for secp160r1 is larger than 2^160, so it requires 161 bits to represent.

The value of a is implicit in the definition of the x_side() function for each curve.

closed 2017-05-21 by kmackay

edogaldo, 2017-05-22:
Thank you Ken but I'm still missing the point for "a": in the code I can see 2 instances of x_side():
- x_side_secp256k1() which computes "x^3 + b" and is used for curve p256k1
- x_side_default() which computes "x^3 - 3x + b" and seems used for all other curves
I can't see a dedicated x_side() for each curve, am I wrong?!

Thanks in advance and best regards, E.