kmackay/micro-ecc #104: Makefile not found

vineeshvs, 2017-05-13:
1. The [link](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v12.0.0%2Flib_crypto.html&cp=4_0_0_3_3_1_2&anchor=lib_crypto_installing says)
`Run make to compile the micro-ecc library.`

But I am  not able to find the Makefile?

2. If I already have a micro-ecc folder in the 'external' directory. Where should I copy the files in this repository (after cloning)

3. Should I copy all the files or just the .c and .h files?

kmackay, 2017-05-21:
I don't think I ever had a Makefile to build the library; but maybe Nordic created their own?

I'm not sure what the 'external' directory is; but I would expect you can just copy the latest version of micro-ecc over your existing files (and maybe update the Makefile if it exists). You need the .c, .h, and .inc files at least.

msembinelli, 2017-06-17:
You are correct @kmackay. In the Nordic nRF5 SDK 12 and above they have distributed makefiles and scripts to clone micro-ecc from github and build it. @vineeshvs you can find these scripts in the Nordic SDK under external/micro-ecc. They have a script called 'build_all.bat' (windows) and 'build_all.sh' (linux). You can just run those instead of make manually to make it easier.

[image]

You must run the scripts/makefiles in place, do not move the folder out of the SDK. Their makefile references a parent makefile buried somewhere deeper within the SDK. It needs to be in a path relative to the makefile.

Hope this helps!

closed 2020-10-07 by kmackay