kmackay/micro-ecc #108: How to add other curves?

edogaldo, 2017-05-23:
Dear @kmackay, I'd like to add some other curves (i.e. secp112r1/2, secp128r1/2, secp160k1, secp192k1, secp224k1 from SEC2 v1.0); optimizations apart, can above curves be implemented just using the default functions?
Would it be possible for you to provide an explanation on how to add new curves?

Thanks in advance and best regards, E.

Thanks in advance and best regards, E.

yaohcn, 2018-05-15:
@edogaldo hello, do you have already add other curves?

kmackay, 2020-10-07:
I wouldn't use 112-bit or 128-bit curves, they are too insecure (and are hard to implement).

It should be possible to implement the other curves you mentioned using the default functions, similar to the existing implementation of secp256k1

closed 2020-10-07 by kmackay