kmackay/micro-ecc #111: uECC_OPTIMIZATION_LEVEL 1 leads to errors

Sossenbinder, 2017-06-14:
Hello,

I tried to set the uECC_OPTIMIZATION_LEVEL to 1, as mentioned in the uECC.h file in order to lower code size.

However, I am left with these errors:

     asm_avr.inc:879:5:  error: invalid lvalue in asm output 3

     __asm__ volatile (

     ^

and

       asm_avr.inc: In function 'uECC_vli_sub':
       asm_avr.inc:892:58: error: 'i' undeclared (first use in this function)

         : "+z" (r), "+x" (left), "+y" (right), [i] "+r" (i),

                                                          ^

Any idea how to fix this?

kmackay, 2017-06-14:
Are you compiling with -O1 or higher?

Sossenbinder, 2017-06-19:
I don't think so, I am currently using the Arduino IDE, and according to the config files no optimiziation flag is given. Is this necessary?

kmackay, 2017-06-20:
-Os should work. I see there is an error.

Sossenbinder, 2017-06-20:
Thanks. I can now compile without any errors.

closed 2017-06-20 by Sossenbinder