kmackay/micro-ecc #116: Wrong signature compared with libsecp256k1

iskradelta, 2017-06-26:
Hi,

I tried to reuse a keypair which was generated by libsecp256k1, and get a different signature when using uECC_sign_deterministic with bearssls sha256. 

This is the private key in base64, "j5fBQFQuKCbZoWqclzNQ/GCgtPcxZ0Fu2OYy95qmVXs=", and public key compressed in base64, "Awa7pBOZx4HHbbtT5J6820maRoFQzmVynWFgxyF6As+a".

uECC_sign_deterministic is called like in the test_ecdsa_deterministic example, only the sha256 init, update and finish functions are pointing/using BearSSL implementations, br_sha256_init, br_sha256_update and br_sha256_out (Ive also verified the problem is not in these hashing functions by hashing several messages and comparing with pythons sha functions.).

Any further idea how to troubleshoot?

kmackay, 2017-06-27:
I don't expect that you would get the same signature value since there are some implementation differences. uECC generates a valid signature that is deterministic, but it won't be necessarily the same value as some other library.

iskradelta, 2017-06-27:
I mean, the generated signature by micro-ecc does not verify on libsecp256k1, while the sig from libsecp256k1, does verify with micro-ecc. 

Should it really be like that, signatures generated by micro-ecc failing on other implementations?

kmackay, 2017-06-28:
Oh. The signatures should verify in both directions. But just because the signature is different doesn't mean that it isn't valid. It's hard for me to tell what's going wrong for you though; maybe the signatures are in different formats?

iskradelta, 2017-06-29:
Ah, you can close the issue again. 

It was only the private key I tried to import which failed. When generating a new keypair with micro-ecc,, saving it, reading it again, those signatures verified good on libsecp256k1. Must have imported the key in wrong format, probably the endianness bit me too. Sorry for the hassle, and thank you for your time and great library!

closed 2017-06-30 by kmackay