kmackay/micro-ecc #129: uECC_shared_secret takes > 10 secs

derchrismakaio, 2017-11-17:
Dear all,
I know this is not an issue with micro-ecc but with my setup. I'm using the static version (with uECC_secp256r1) for a thumb v1 Cortex m0 and the shared secret computation takes really long. 
Here are my compiler options:
-mcpu=cortex-m0 -mthumb -std=c99 -O3 -Os

uECC_SQUARE_FUNC is 1
uECC_ASM is uECC_asm_fast

Can anyone point me to the right direction where something obviously could go wrong?

derchrismakaio, 2017-11-17:
I did some logging and all the time is spent in EccPoint_mult

kokke, 2017-12-04:
Compiler options of `-O3 -Os` seem contradictory? You would often define either `-O3` or `-Os` to optimize for either performance or size. Try removing `-Os` at the cost of possibly expanding the code foot print.

kmackay, 2020-10-07:
This is expected for slow microcontrollers; ECC operations in general can take several seconds on such devices.

closed 2020-10-07 by kmackay