kmackay/micro-ecc #131: Signature Format

G1H, 2018-02-15:
Hi all,

I'm trying to use micro-ecc to verify a signature generated by another library producing ASN1 formatted signatures. However, I'm struggling to convert it into a format compatible with uECC.

Investigating the signature generated by uECC (using secp256k1 curve, sha256 and the static version of uECC), i get 64 bytes signature with the first 32bytes staying constant (only changing with a different curve) while the last 32 bytes change when using different keys. 

Is this expected behavior? Am I able to discard the first 32 bytes of the signature when transmitting to the receiver?

Can anybody provide information about the signature format used by micro-ecc and how to produce it with other libraries (e.g. openssl)?

Thanks,
GH

G1H, 2018-02-16:
Ok, just figured out that some people had similar problems when going through the closed issues. 
I'll give their solutions a try.

Still, I would like to understand why first 32 bytes stay constant when creating signatures with the same curve but different keys.

kmackay, 2020-10-08:
If you're using uECC_sign_deterministic(), then I think the first 32 bytes would always be the same for a given message, regardless of key (since k is constant, derived from the message). If you're using uECC_sign() then it's a sign that your RNG is broken (always produces the same value).

closed 2020-10-08 by kmackay