kmackay/micro-ecc #133: asm_avr.inc undefine reference

Sinitax, 2018-03-10:
When compiling a modified version of the example sketch from the master branch i get this error:
```
C:\Users\Sinitax\AppData\Local\Temp\ccWK0HZp.ltrans0.ltrans.o: In function `vli_modSub_fast.lto_priv.30':
C:\Program Files (x86)\Arduino\libraries\micro-ecc-static/asm_avr.inc:21430: undefined reference to `curve_p'
C:\Program Files (x86)\Arduino\libraries\micro-ecc-static/asm_avr.inc:21430: undefined reference to `curve_p'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Nano.
```

This is the code from the sketch:
```
#include <uECC.h>

#define uECC_ASM uECC_asm_small
#define ECC_SQUARE_FUNC 0
#define uECC_CURVE uECC_secp256k1

extern "C" {

static int RNG(uint8_t *dest, unsigned size) {
  // Use the least-significant bits from the ADC for an unconnected pin (or connected to a source of 
  // random noise). This can take a long time to generate random data if the result of analogRead(0) 
  // doesn't change very frequently.
  while (size) {
    uint8_t val = 0;
    for (unsigned i = 0; i < 8; ++i) {
      int init = analogRead(0);
      int count = 0;
      while (analogRead(0) == init) {
        ++count;
      }
      
      if (count == 0) {
         val = (val << 1) | (init & 0x01);
      } else {
         val = (val << 1) | (count & 0x01);
      }
    }
    *dest = val;
    ++dest;
    --size;
  }
  // NOTE: it would be a good idea to hash the resulting random data using SHA-256 or similar.
  return 1;
}

}  // extern "C"

void setup() {
  Serial.begin(115200);
  Serial.print("Testing ecc\n");
  uECC_set_rng(&RNG);
}

void loop() {
  uint8_t private1[21];
  uint8_t private2[21];
  
  uint8_t public1[40];
  uint8_t public2[40];
  
  uint8_t secret1[20];
  uint8_t secret2[20];
  
  unsigned long a = millis();
  uECC_make_key(public1, private1);
  unsigned long b = millis();
  
  Serial.print("Made key 1 in "); Serial.println(b-a);
  a = millis();
  uECC_make_key(public2, private2);
  b = millis();
  Serial.print("Made key 2 in "); Serial.println(b-a);

  a = millis();
  int r = uECC_shared_secret(public2, private1, secret1);
  b = millis();
  Serial.print("Shared secret 1 in "); Serial.println(b-a);
  if (!r) {
    Serial.print("shared_secret() failed (1)\n");
    return;
  }

  a = millis();
  r = uECC_shared_secret(public1, private2, secret2);
  b = millis();
  Serial.print("Shared secret 2 in "); Serial.println(b-a);
  if (!r) {
    Serial.print("shared_secret() failed (2)\n");
    return;
  }
    
  if (memcmp(secret1, secret2, 20) != 0) {
    Serial.print("Shared secrets are not identical!\n");
  } else {
    Serial.print("Shared secrets are identical\n");
  }
}
```

mryalamanchi, 2018-07-30:
@Sinitax, there are 2 errors in your code:

1) You haven't initialized the curve using uECC_curve_t. 
`const struct uECC_Curve_t * curve = uECC_secp256k1();`

2)  You haven't mentioned the curve to be used in uECC_make_key and uECC_shared_secret function calls. 
```
uECC_make_key(publicN, privateN,**curve**);
int r = uECC_shared_secret(publicN, privateN, secretN, **curve**);
```

Sinitax, 2018-07-30:
Sorry but since this issue is 5 months old I can't test your code myself. I think that at the time I was using a forked version of micro-ecc that forgot to update the examples with uECC_Curve_t and instead used the #define syntax from an older release, which caused this error. Thanks for your reply, I'm sure it will help many with the same problem!

closed 2018-07-30 by Sinitax