kmackay/micro-ecc #136: Crash on Node-MCU V3 (ESP8266) when using uECC_verify many times in a row

peterwilli, 2018-05-10:
Hi there,

I have a very weird crash when using `uECC_verify` often (after hours of debugging, it seems that function is the culprit).

I have a Node-MCU v3 (with Arduino firmware), which I program using the platformio-ide on Atom.
The library itself works fine up to the point it starts to crash. I couldn't find any memory leaks or anything.

I've added a test package with a clean project where 1 is the latest micro-ecc library with no modifications  and just the test example on it. The other is the same but with the static version. Both crash at some point (to me after around 5 tries).

Just load it up your board and watch it crash to reproduce.
If you need more info let me know :)

[ECCTests.zip](https://github.com/kmackay/micro-ecc/files/1991324/ECCTests.zip)

abdoutech93, 2018-05-30:
Did you tried the uECC_make_key(public2, private2, curve); !? 
It crash when i call this function !! I test it on Node-MCU v2 with arduino firmware using platformio-ide on VS Code.

peterwilli, 2018-05-31:
@abdoutech93 do you have a RNG function set? I found that the RNG in example also crashes my ESP. After this and the delay it works perfectly over here.
Here is a copy of my RNG:

```
extern "C" {

int RNG(uint8_t *p_dest, unsigned p_size) {
  uint16_t i = 0;
  while (i < p_size) {
    uint32_t v = RANDOM_REG32;
    uint32_t l_amount = min(p_size - i, sizeof(uint32_t));
    memcpy(&p_dest[i], &v, l_amount);
    i += l_amount;
  }
  char hash[SHA256_BLOCK_SIZE];
  Sha256 sha256Instance = Sha256();
  sha256Instance.update((const BYTE *)p_dest, p_size);
  sha256Instance.final((BYTE *)hash);
  memcpy(p_dest, hash, MIN(p_size, SHA256_BLOCK_SIZE));
  return 1;
}

} // extern "C"
```

The RNG is dependent on SHA256 over here: https://github.com/B-Con/crypto-algorithms/blob/master/sha256.c

abdoutech93, 2018-06-01:
Thank you so much, this made my day. please did this sha256 hashing lib is existing on platformio !?

peterwilli, 2018-06-04:
@abdoutech93 I have it included next to my other files in `src/` (see link above, grab the .c and .h file and it works instantly)

closed 2020-10-07 by kmackay