kmackay/micro-ecc #137: Can I use on Linux C?

jcptk677, 2018-05-23:
Hello, I tried to use 'micro-ecc-master/test/test_ecdh.c' on LINUX C. So, I moved that file into 'micro-ecc-master/' and then typed "gcc -o test_ecdh test_ecdh.c". But I could not make exe file with an error msg below.

/tmp/ccwq3btr.o: In function `main':
test_ecdh.c:(.text+0x158): undefined reference to `uECC_secp160r1'
test_ecdh.c:(.text+0x17a): undefined reference to `uECC_secp192r1'
test_ecdh.c:(.text+0x19c): undefined reference to `uECC_secp224r1'
test_ecdh.c:(.text+0x1be): undefined reference to `uECC_secp256r1'
test_ecdh.c:(.text+0x1e0): undefined reference to `uECC_secp256k1'
test_ecdh.c:(.text+0x258): undefined reference to `uECC_make_key'
test_ecdh.c:(.text+0x282): undefined reference to `uECC_make_key'
test_ecdh.c:(.text+0x2c4): undefined reference to `uECC_shared_secret'
test_ecdh.c:(.text+0x309): undefined reference to `uECC_shared_secret'
collect2: error: ld returned 1 exit status

can you help me?...

z1Y2x, 2018-05-23:
Missing symbols be part of uECC source. Try „gcc -o test_ecdh test_ecdh.c ../uECC.c“

zouzhihui, 2019-05-07:
@kmackay 
micro-ecc is nice~
it can run on linux/unix?

zouzhihui, 2019-05-07:
@z1Y2x 

can you resolve this issue? I want to run on linux/unix(centos7)?

jcptk677, 2019-05-07:
I solved this issue.

zouzhihui, 2019-05-07:
> […](#)

@jcptk677 

How to solve, can you tell me ?

jcptk677, 2019-05-07:
I used this command.

gcc -o test_ecdh test_ecdh.c ../uECC.c

kmackay, 2020-10-07:
Yes, you need to build against uECC.c.

closed 2020-10-07 by kmackay