kmackay/micro-ecc #148: Example ecc_test.ino not compiling on Arduino 1.8.8

GitMoDu, 2019-02-25:
I get the following message, on the RNG function declaration.

> expected unqualified-id before string constant

I was able to compile the example, by removing extern "C" section.

Hans-MartinHannibalLauridsen, 2019-10-03:
What device are you compiling for? I'm experiencing the pro Micro being unresponcive for New uploads when using this this hack. The only way i have found to revive it is to burn it with at New bootloader.

GitMoDu, 2019-10-04:
Wow, that's extreme. I've tested on ATtiny85, ATmega328p and STM32F1, ECC is working properly.

PanosChtz, 2020-01-15:
I'm having the exact same error. My board is ATmega328p and Arduino 1.8.9. Removing extern "C" section as suggested above doesn't help me.

PanosChtz, 2020-01-17:
Solution for the error: put following code right before extern "C" {

`extern "C" {
static int RNG(uint8_t *dest, unsigned size);
}`

Hans-MartinHannibalLauridsen, 2020-01-31:
Not sure this is the way to go about, but i'm on version 1.8.9 and seems to have luck doing this:
`
#include <uECC.h>

  static int RNG(uint8_t *dest, unsigned size) {
    // Use the least-significant bits from the ADC for an unconnected pin (or connected to a source of 
    // random noise). This can take a long time to generate random data if the result of analogRead(0) 
    // doesn't change very frequently.
    while (size) {
      uint8_t val = 0;
      for (unsigned i = 0; i < 8; ++i) {
        int init = analogRead(0);
        int count = 0;
        while (analogRead(0) == init) {
          ++count;
        }
        
        if (count == 0) {
           val = (val << 1) | (init & 0x01);
        } else {
           val = (val << 1) | (count & 0x01);
        }
      }
      *dest = val;
      ++dest;
      --size;
    }
    // NOTE: it would be a good idea to hash the resulting random data using SHA-256 or similar.
    return 1;
  }

void setup() {
  Serial.begin(115200);
  Serial.print("Testing ecc\n");
  delay(15000); //Delay 15 sec to avoid serial interference 
  uECC_set_rng(&RNG);
}
`
The uploading problems I had with the atmega32u2 seem to be related with the nature of how serial communication is handled om the module (maybe), hence the delay of 15sec to make sure that the serial isn't busy when attempting to upload a new sketch.. (or so the web would have me believe).

closed 2020-10-07 by kmackay