kmackay/micro-ecc #151: Invalid public keys accepted by uECC_shared_secret()

tim-dorcey, 2019-03-12:
Header file says uECC_valid_public_key() is not required by other functions, except as potential optimization.  However, there are known attacks on ECDH using invalid public keys.  I created an invalid public key (for sec256k1) by setting x = curve->p + 1, and then computing y coordinate with uECC_decompress().  uECC_shared_secret() created a shared secret without error.

I am not sure if there are any known attacks based on this particular kind of invalid key.

kmackay, 2020-10-07:
Callers may call uECC_valid_public_key() to verify the public key prior to calling uECC_shared_secret(). I don't want to incur the code size/time cost of always calling uECC_valid_public_key() within uECC_shared_secret(), since callers may already know it is valid.

closed 2020-10-07 by kmackay

tim-dorcey, 2020-10-08:
I agree.  Was just suggesting comment in header file should change.  Great code!