kmackay/micro-ecc #165: Not constant time secret key handling

gmaxwell, 2020-01-28:
The documentation states that this in intended to be 'Resistant to known side-channel attacks', but I believe this claim is currently mistaken. I don't envy the effort it takes to make constant time code which is flexible across many curves and systems as small as 8 bits.

Here are a few examples issues I see (reviewing from the perspective of gcc9 on x86_64 on linux, w/ secp256k1):

uECC.c:877         nb = !uECC_vli_testBit(scalar, i);
uECC.c:878         XYcZ_addC(Rx[1 - nb], Ry[1 - nb], Rx[nb], Ry[nb], curve);

Array accesses have variable time based on the position. See http://www.tau.ac.il/~tromer/papers/cache.pdf

(Any they still do even if an extraordinary effort to made the accesses cache-line regular, which this does not, See: https://cryptojedi.org/peter/data/chesrump-20130822.pdf)

Also, while computing uECC_vli_modMult at uECC.c:1290 (s = r*d)

uECC.c:592 uECC_word_t diff = v[index][i] - mod_multiple[i] - borrow;

Index is secret.

uECC.c:588    for (index = 1; shift >= 0; --shift) {

Shift is secret, but the loop count depends on it.

And during  uECC_vli_modAdd(s, tmp, s, curve->n, num_n_words); /* s = e + r*d */ at uECC:1293 

uECC.c:541    if (carry || uECC_vli_cmp_unsafe(mod, result, num_words) != 1) {

Which, as the name suggests, uses a very not constant time comparison. :)

Though some of these leaks can seem pretty small-- It takes only an incredibly small leak to blow security (I apologize if this is something you already know.).

For example, if the leak allows an attacker to filter a bunch of signatures to select ones with k a bit or two smaller they can extract the private key with from only a few hundred filtered signatures. If you haven't seen it, check out: https://eprint.iacr.org/2014/161.pdf

kmackay, 2020-10-07:
k is regularized, so it always has the same number of bits.

kmackay, 2020-10-07:
Regarding uECC_vli_mmod(), the index and shift are not secret. They only depend on the number of bits in the divisor (n), which is public.

kmackay, 2020-10-07:
I agree that there may be timing leaks due to caching effects. Note that this library was intended for use on microcontrollers, which often don't have caching, and so timing leaks may be harder to achieve on those devices. At any rate, I'm not claiming it is invulnerable to timing attacks, just resistant to them

closed 2020-10-07 by kmackay