kmackay/micro-ecc #168: Potentially SPA-vulnerability

pmvr, 2020-04-25:
Though the implementation of the scalar elliptic point multiplication algorithm for ECDSA is timing invariant, it does not use a randomized z-coordinate of the elliptic curve points, which make it potentially vulnerable against SPA. Line 1258 of uECC.c executes

`EccPoint_mult(p, curve->G, k2[!carry], 0, num_n_bits + 1, curve);`

The 4th parameter is initial_Z which, if set to 0, sets the z-coordinate to 1 when line 874 executes XYcZ_initial_double. Since the ephemeral key is processed bitwise, two elliptic curve point multiplications with the same sequence of leading bits lead to the same computations on the device, which could generate similar power consumption traces.

pmvr, 2020-06-12:
See [here](https://pmvr.github.io/) for more details.

kmackay, 2020-10-07:
Thanks. Added random initial Z value.

closed 2020-10-07 by kmackay

javierMolinero, 2024-01-02:
Hi, one question about this fix. It iss supposed to solve the vulnerability described in https://nvd.nist.gov/vuln/detail/CVE-2020-27209 ? The vulnerability describe than the base point G should be randomized to protect the operation against SDA. When randomizing initial_Z, it seems to randomize then G, then I understand that it could be valid for adding countermeasures against SDA described on it, is right? Thanks in advance