kmackay/micro-ecc #172: uECC_verify always return 0

mwm93, 2020-07-08:
I try to sign a hash with uECC. 

I use the privkey
0x17 0x16 0x15 0x14 0x13 0x12 0x11 0x10 0xf 0xe 0xd 0xc 0xb 0xa 0x9 0x8 0x7 0x6 0x5 0x4 0x3 0x2 0x1 0x0

And the pubkey 
0xc2 0x26 0x6c 0xf0 0x42 0x1b 0xd7 0x2f 0x80 0x96 0x75 0x4f 0xa2 0x97 0x2a 0x62 0x6a 0xcd 0x3e 0x9a 0x42 0xe8 0x3d 0x89 0xbf 0x1 0x53 0x58 0x9 0x17 0x19 0x14 0xee 0xd6 0x6a 0x14 0x88 0xd5 0x3 0x6d 0xf7 0x30 0x31 0xf 0xbc 0x93 0xd6 0x68

And the sha256: 
0x63 0xce 0x66 0x9b 0x9e 0x43 0xcf 0xf 0xe5 0xa5 0x7d 0x95 0xfb 0x22 0x71 0xb3 0x6a 0x85 0x16 0x9f 0x4f 0x45 0x5f 0xe7 0xdc 0x2b 0xec 0x53 0x5b 0x51 0xd8 0x2e

I sign and get the following result:
0x44 0xd2 0x6c 0x7 0x30 0xc2 0x79 0x83 0x42 0x9c 0x8a 0xcd 0xd1 0x62 0x29 0x2c 0x12 0x1b 0x8 0xc7 0x9 0xfb 0x6c 0xed 0x73 0x24 0xa2 0x92 0xfb 0xe 0xbd 0xf4 0x61 0xed 0x32 0x1c 0x32 0x43 0xee 0xa6 0xf3 0x6e 0x97 0x5f 0xd4 0x9b 0x14 0xfc

I than call 

    int res = uECC_verify(pubkey,
                md,
                SHA256_DIGEST_LENGTH,
                signatureRaw,
                uECC_secp192r1());
                
with the corresponding parameters and it always return 0. I tried it with both static as well as master branch. No one of them worked.

kmackay, 2020-10-07:
It may be helpful to provide the code you are using to generate the keypair and signature. Note that you can see test/test_ecdsa.c for an example of signing and verifying.

kmackay, 2020-10-08:
I also posted some example code in #173 for verifying an existing signature (from OpenSSL).

closed 2020-10-08 by kmackay