kmackay/micro-ecc #173: Failure to verify against openssl

rmincher, 2020-07-26:
I built the library and am running under osx.   I am using the secp256r1 curve.

If I run the example to sign / verify using uECC it will verify.

What fails is when I try to use uECC to verify a signature generated by openssl, or use openssl to verify using a signature generated by uECC.

I have generated a private key with openssl and verified that uECC_make_public_key() will generate the same public key that openssl is getting so I am confident the curve is matching.

My key generation, signing and verification script:
```#openssl ecparam -name prime256v1 -genkey -noout -out private.pem
openssl ec -in private.pem -pubout -out public.pem  #-conv_form uncompressed
openssl ec -in private.pem -text
touch msg.txt | echo -n "aaa" > msg.txt
openssl dgst -sha256 -out msg.digest.txt msg.txt
cat msg.digest.txt
openssl dgst -sha256 -sign private.pem -out msg.signature.txt msg.digest.txt
openssl asn1parse -inform DER -in msg.signature.txt
python3 dumpsig.py <msg.signature.txt
openssl dgst -sha256 -hex -verify public.pem -signature msg.signature.txt msg.digest.txt
```

And output of script:
```read EC key
writing EC key
read EC key
Private-Key: (256 bit)
priv:
    5f:ef:5f:e5:62:7c:ff:60:56:87:3e:64:87:32:72:
    c3:ed:8c:08:c4:b6:01:b1:3c:36:c0:db:c2:bb:8d:
    24:fc
pub: 
    04:f1:bf:ca:c9:3c:ff:50:71:4f:7d:cc:1c:23:e7:
    03:02:d1:81:8e:88:52:20:43:01:8a:3e:68:98:78:
    9b:8a:9f:48:a2:a8:43:f4:2a:e3:3c:5a:da:d8:90:
    bc:5d:f3:87:98:64:fb:b8:ea:e7:ea:95:5e:39:84:
    3a:85:44:a3:49
ASN1 OID: prime256v1
NIST CURVE: P-256
writing EC key
-----BEGIN EC PRIVATE KEY-----
MHcCAQEEIF/vX+VifP9gVoc+ZIcycsPtjAjEtgGxPDbA28K7jST8oAoGCCqGSM49
AwEHoUQDQgAE8b/KyTz/UHFPfcwcI+cDAtGBjohSIEMBij5omHibip9IoqhD9Crj
PFra2JC8XfOHmGT7uOrn6pVeOYQ6hUSjSQ==
-----END EC PRIVATE KEY-----
SHA256(msg.txt)= 9834876dcfb05cb167a5c24953eba58c4ac89b1adf57f28f2f9d09af107ee8f0
    0:d=0  hl=2 l=  68 cons: SEQUENCE          
    2:d=1  hl=2 l=  32 prim: INTEGER           :58CF628E0597B8AE32BD8199C6E7DF0192A50CF3AAC07F66769286FB00F45815
   36:d=1  hl=2 l=  32 prim: INTEGER           :78235F960F70E5641B6F1CCE080F897F20722541BDD6A81FFA5DF3C2B130FAEB
signature(70) = 3044022058cf628e0597b8ae32bd8199c6e7df0192a50cf3aac07f66769286fb00f45815022078235f960f70e5641b6f1cce080f897f20722541bdd6a81ffa5df3c2b130faeb
Verified OK
```
My uECC test code is simply:
```
char sig[] = "58CF628E0597B8AE32BD8199C6E7DF0192A50CF3AAC07F66769286FB00F4581578235F960F70E5641B6F1CCE080F897F20722541BDD6A81FFA5DF3C2B130FAEB";
char priv[] = "5fef5fe5627cff6056873e64873272c3ed8c08c4b601b13c36c0dbc2bb8d24fc";

void uECC_test(void)
{
  int r;
  const struct uECC_Curve_t * curve = uECC_secp256r1();

  uint8_t private1[32];
  uint8_t public1[64];

 uint8_t signature[64];

  uint8_t hash[32] = {0x98, 0x34, 0x87, 0x6D, 0xCF, 0xB0, 0x5C, 0xB1, 0x67, 0xA5, 0xC2, 0x49, 0x53, 0xEB, 0xA5, 0x8C, 0x4A, 0xC8, 0x9B, 0x1A, 0xDF, 0x57, 0xF2, 0x8F, 0x2F, 0x9D, 0x09, 0xAF, 0x10, 0x7E, 0xE8, 0xF0};
  dump("hash", hash, sizeof(hash));

  strtobytes(priv, private1);
  dump("priv", private1, sizeof(private1));
  uECC_compute_public_key(private1, public1, curve);
  dump("pub ", public1, sizeof(public1));

  memset(signature, 0xDB, sizeof(signature));
  r =  uECC_sign(private1, hash, sizeof(hash), signature, curve);

//  strtobytes(sig, signature);  // if to override uECC signature with openssl signature
  dump("sig ", signature, sizeof(signature));

  r = uECC_verify(public1, hash, sizeof(hash), signature, curve);
  printf("verify result = %d\n", r);
}
```
When I run that using the signature generated by uECC, it works and I get:
```hash (32): 9834876dcfb05cb167a5c24953eba58c4ac89b1adf57f28f2f9d09af107ee8f0
priv (32): 5fef5fe5627cff6056873e64873272c3ed8c08c4b601b13c36c0dbc2bb8d24fc
pub  (64): f1bfcac93cff50714f7dcc1c23e70302d1818e88522043018a3e6898789b8a9f48a2a843f42ae33c5adad890bc5df3879864fbb8eae7ea955e39843a8544a349
sig  (64): 9fff2fceeb0015300030b78b5c930340408c2145390980e3b4bdff613eb4658e9690d1c9c970bdc1f16fee457d550ede4758df8ea3e24dfb3dee2d6481f1cf98
verify result = 1
```

If I override the signature generated by uECC (uncomment the strtobytes call), I get:
```hash (32): 9834876dcfb05cb167a5c24953eba58c4ac89b1adf57f28f2f9d09af107ee8f0
priv (32): 5fef5fe5627cff6056873e64873272c3ed8c08c4b601b13c36c0dbc2bb8d24fc
pub  (64): f1bfcac93cff50714f7dcc1c23e70302d1818e88522043018a3e6898789b8a9f48a2a843f42ae33c5adad890bc5df3879864fbb8eae7ea955e39843a8544a349
sig  (64): 58cf628e0597b8ae32bd8199c6e7df0192a50cf3aac07f66769286fb00f4581578235f960f70e5641b6f1cce080f897f20722541bdd6a81ffa5df3c2b130faeb
verify result = 0
```
I have the same issue if I take the 64-byte signature generated by uECC above, converting it to asn1/DER format and trying to get openssl to verify the signature.
```sig  (64): 9fff2fceeb0015300030b78b5c930340408c2145390980e3b4bdff613eb4658e9690d1c9c970bdc1f16fee457d550ede4758df8ea3e24dfb3dee2d6481f1cf98

$ python3 mksig.py 9fff2fceeb0015300030b78b5c930340408c2145390980e3b4bdff613eb4658e9690d1c9c970bdc1f16fee457d550ede4758df8ea3e24dfb3dee2d6481f1cf98 >junk
$ openssl asn1parse -inform DER -in junk
    0:d=0  hl=2 l=  70 cons: SEQUENCE          
    2:d=1  hl=2 l=  33 prim: INTEGER           :9FFF2FCEEB0015300030B78B5C930340408C2145390980E3B4BDFF613EB4658E
   37:d=1  hl=2 l=  33 prim: INTEGER           :9690D1C9C970BDC1F16FEE457D550EDE4758DF8EA3E24DFB3DEE2D6481F1CF98
 $ openssl dgst -sha256 -hex -verify public.pem -signature junk msg.digest.txt
Verification Failure
```
Is there a reason that this is not working for me?  I understand that this is for embedded systems and that is where I plan to use it, but I wanted to at least sanity check it on a more friendly platform before investing more time with it.

Thank you.

mwm93, 2020-07-27:
I have simmiliar Problems. 

I tried to generate a pubkey with uECC and OpenSSL from the same private key. 

I recieved: 

```
uECC:
Privkey in Hex: [0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9 0xa 0xb 0xc 0xd 0xe 0xf 0x10 0x11 0x12 0x13 0x14 0x15 0x16 0x17 0x18 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 
Pubkey in Hex: [0xb4 0xd0 0x1f 0xb1 0x85 0xdc 0x8a 0x7e 0x22 0xfb 0x67 0x77 0xcd 0x73 0xaa 0x81 0x3b 0xde 0x82 0xcb 0x4a 0xb0 0x62 0xb1 0xe2 0x30 0x9d 0xc3 0xb4 0xc4 0x5c 0xfd 0xf 0x3 0xb0 0x2a 0xf4 0xd2 0xa4 0x6f 0x1b 0xea 0x55 0xd6 0xb 0x2e 0x83 0x7a 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 

OpenSSL (uncompressed):
Privkey in Hex: [0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9 0xa 0xb 0xc 0xd 0xe 0xf 0x10 0x11 0x12 0x13 0x14 0x15 0x16 0x17 0x18 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 
Pubkey in Hex: [0x04, 0x9c 0x35 0xbc 0xc7 0x63 0x4b 0x6d 0x6c 0x76 0xb1 0x1a 0x89 0x8e 0x83 0xea 0xc5 0x9b 0x88 0x8b 0x1a 0x13 0x71 0xfe 0xd0 0xc1 0x8c 0x0 0x1d 0x66 0x44 0x6d 0x9c 0x20 0x4b 0x40 0xf4 0x52 0xd6 0xa7 0xbc 0x39 0x46 0x65 0xbf 0x8 0x97 0x34 0xc7 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 

OpenSSL hybrid
Privkey in Hex: [0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9 0xa 0xb 0xc 0xd 0xe 0xf 0x10 0x11 0x12 0x13 0x14 0x15 0x16 0x17 0x18 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 
Pubkey in Hex: [0x7 0x9c 0x35 0xbc 0xc7 0x63 0x4b 0x6d 0x6c 0x76 0xb1 0x1a 0x89 0x8e 0x83 0xea 0xc5 0x9b 0x88 0x8b 0x1a 0x13 0x71 0xfe 0xd0 0xc1 0x8c 0x0 0x1d 0x66 0x44 0x6d 0x9c 0x20 0x4b 0x40 0xf4 0x52 0xd6 0xa7 0xbc 0x39 0x46 0x65 0xbf 0x8 0x97 0x34 0xc7 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 

OpenSSL (compressed)
Privkey in Hex: [0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9 0xa 0xb 0xc 0xd 0xe 0xf 0x10 0x11 0x12 0x13 0x14 0x15 0x16 0x17 0x18 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 
Pubkey in Hex: [0x3 0x9c 0x35 0xbc 0xc7 0x63 0x4b 0x6d 0x6c 0x76 0xb1 0x1a 0x89 0x8e 0x83 0xea 0xc5 0x9b 0x88 0x8b 0x1a 0x13 0x71 0xfe 0xd0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 

uECC without Endian-Convertation:
Privkey in Hex: [0x18 0x17 0x16 0x15 0x14 0x13 0x12 0x11 0x10 0xf 0xe 0xd 0xc 0xb 0xa 0x9 0x8 0x7 0x6 0x5 0x4 0x3 0x2 0x1 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 
Pubkey in Hex: [0xb1 0x62 0xb0 0x4a 0xcb 0x82 0xde 0x3b 0x81 0xaa 0x73 0xcd 0x77 0x67 0xfb 0x22 0x7e 0x8a 0xdc 0x85 0xb1 0x1f 0xd0 0xb4 0x7a 0x83 0x2e 0xb 0xd6 0x55 0xea 0x1b 0x6f 0xa4 0xd2 0xf4 0x2a 0xb0 0x3 0xf 0xfd 0x5c 0xc4 0xb4 0xc3 0x9d 0x30 0xe2 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 

Switching to another Curve, secp256k1:

uECC uECC_VLI_NATIVE_LITTLE_ENDIAN=0
Privkey in Hex: [0x20 0x1f 0x1e 0x1d 0x1c 0x1b 0x1a 0x19 0x18 0x17 0x16 0x15 0x14 0x13 0x12 0x11 0x10 0xf 0xe 0xd 0xc 0xb 0xa 0x9 0x8 0x7 0x6 0x5 0x4 0x3 0x2 0x1 
Pubkey in Hex: [0x84 0xbb 0x7 0x71 0x42 0xc3 0x1 0xd4 0x71 0xa3 0x3a 0x99 0x5b 0x22 0x9 0xdb 0xe3 0x78 0x89 0xd0 0x1b 0xe0 0x31 0xe6 0xb0 0x9d 0xdc 0x65 0x73 0x1b 0x19 0x62 0xf2 0x4d 0x6f 0x7d 0xaa 0x43 0x1 0x7d 0x27 0x57 0x5f 0x25 0x4c 0x4f 0xcc 0x74 0x96 0x44 0x75 0x6c 0x3f 0x91 0xa1 0x50 0x58 0xff 0xdb 0x34 0xf4 0x66 0x15 0xe6 

uECC uECC_VLI_NATIVE_LITTLE_ENDIAN=1
Privkey in Hex: [0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9 0xa 0xb 0xc 0xd 0xe 0xf 0x10 0x11 0x12 0x13 0x14 0x15 0x16 0x17 0x18 0x19 0x1a 0x1b 0x1c 0x1d 0x1e 0x1f 0x20 
Pubkey in Hex: [0x62 0x19 0x1b 0x73 0x65 0xdc 0x9d 0xb0 0xe6 0x31 0xe0 0x1b 0xd0 0x89 0x78 0xe3 0xdb 0x9 0x22 0x5b 0x99 0x3a 0xa3 0x71 0xd4 0x1 0xc3 0x42 0x71 0x7 0xbb 0x84 0xe6 0x15 0x66 0xf4 0x34 0xdb 0xff 0x58 0x50 0xa1 0x91 0x3f 0x6c 0x75 0x44 0x96 0x74 0xcc 0x4f 0x4c 0x25 0x5f 0x57 0x27 0x7d 0x1 0x43 0xaa 0x7d 0x6f 0x4d 0xf2 

OpenSSL, POINT_CONVERSION_COMPRESSED
Privkey in Hex: [0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9 0xa 0xb 0xc 0xd 0xe 0xf 0x10 0x11 0x12 0x13 0x14 0x15 0x16 0x17 0x18 0x19 0x1a 0x1b 0x1c 0x1d 0x1e 0x1f 0x20 
Pubkey in Hex: [0x2 0x84 0xbf 0x75 0x62 0x26 0x2b 0xbd 0x69 0x40 0x8 0x57 0x48 0xf3 0xbe 0x6a 0xfa 0x52 0xae 0x31 0x71 0x55 0x18 0x1e 0xce 0x31 0xb6 0x63 0x51 0xcc 0xff 0xa4 0xb0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 
```

Everytime the same. Propably uECC has serveral Bugs, see 

- https://github.com/kmackay/micro-ecc/issues/172
- https://github.com/kmackay/micro-ecc/issues/168
- https://github.com/kmackay/micro-ecc/issues/167

and not to be used.

motters, 2020-07-28:
I'm having the same issue, following a similar procedure as @rmincher.

There seems to be a lot of openssl issues / questions. We could do with a simple python script in the repo to demo everything working micro-ecc <> openssl wise.

mwm93, 2020-07-29:
That would be an idea, but i think we need to use another libary (which one?), because in a lot of cases, see the issues, uECC seems to don't work correctly.

motters, 2020-07-29:
Currently I've switched to mbedtls while this is solved. 

Micro-ecc is smaller flash usage wise (~2kb) so i still want to revert back.

rmincher, 2020-07-29:
@motters Is this in a bare metal or FreeRTOS environment?  I am curious as to how few mbed files need to be added to a project to make this work.  I only need ECDSA, ECDH; don't need all of TLS.

motters, 2020-07-30:
@rmincher bare metal bootloader. I'm in the same situation atm, stripping down mbedtls to just ECDSA. Hence why i really want to get micro-ecc working.

I'm pretty sure this is doable as as mbed do it for their lora ota bootloader aka no rtos, small size see https://github.com/janjongboom/mbed-lorawan-update-client/blob/master/crypto/source/FragmentationEcdsaVerify.cpp

motters, 2020-07-30:
Looks like Nordic uses[ micro-ecc as a backend](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v13.0.0%2Flib_crypto.html). When using openssl they state the following:

[image]

rmincher, 2020-07-30:
Tried setting all of the various options in uECC.h including platform to other, optimization level to 0, endianness.
Tried byte-swapping public key, hash, signature, signature R, signature S.

I think BLE uses little-endian for some reason, but "network" order is big-endian and that should be the default for both OpenSSL and the library (unless that endian flag is set to 1).  

I'll have to try running it on an ARM; maybe there is an issue with x86 or osx.  I can believe that few use it on this platform, but someone must have tried since it seems to try to support it.

motters, 2020-07-30:
I'm running the library on a Cortex M0+ and have the same problem as yourself.

rmincher, 2020-08-03:
Found this:  https://github.com/intel/tinycrypt

It appears to be based on micro-ecc and includes test vectors that include the term "NIST".  I didn't check the CAVP vectors to see if they are truly a match.  The mbedtls code appears to be based on the intel derivative.  This is more stand-alone than the mbed code.  I just started playing with it and haven't verified it with openssl yet, but am working on it.

opi-smccoole, 2020-08-05:
@rmincher Have you had any luck verifying if the mbedtls or mainline tinycrypt derivatives work properly with OpenSSL keys?  They seem to all pretty much implement the same algorithm for scalar multiplication (Montgomery Ladder co-Z) that may have the same issue as the code here for OpenSSL keys on curves like secp256r1.

rmincher, 2020-08-05:
@opi-smccoole I have not had luck with OpenSSL. I did (quite by accident) notice that the first of the ECDSA test vectors contained in the Intel code was included in some source code from ST Micro to demonstrate how the PKA module of their STM32L5 chip works. Leading me to actually locating that vector in the CAVP (NIST) test vectors for ECDSA.  So I knew the vector must be correct. I have not verified, but I suspect that the Intel test vectors are really all of the P-256, SHA-256 vectors that CAVP published.

Using the code shown in the intel test code to convert the key endianness (which I believe is buried inside of this repo's API) I was able to write a small stand-alone case that showed that I could verify the signature (no great accomplishment since all of the intel vectors passed in their test code).

Last night I decided to try the pycryptodome python module to verify the vector and it worked the first time I ran the code.  I haven't figured out how to deterministically sign using the python module yet so I can't duplicate their sign yet. I can only assume pilot error on OpenSSL and I need to find someone who lives OpenSSL to help with that.

Bottom line, I believe that the Intel code works and I assume this code works but you can check this yourself by running this NIST vector through it (I no longer have this set up to build):

```[P-256,SHA-256]
Msg = 44acf6b7e36c1342c2c5897204fe09504e1e2efb1a900377dbc4e7a6a133ec56
d = 519b423d715f8b581f4fa8ee59f4771a5b44c8130b4e3eacca54a56dda72b464
Qx = 1ccbe91c075fc7f4f033bfa248db8fccd3565de94bbfb12f3c59ff46c271bf83
Qy = ce4014c68811f9a21a1fdb2c0e6113e06db7ca93b7404e78dc7ccd5ca89a4ca9
k = 94a1bbb14b906a61a280f245f9e93c7f3b4a6247824f5d33b9670787642a68de
R = f3ac8061b514795b8843e3d6629527ed2afd6b1f6a555a7acabb5e6f79c8c2ac
S = 8bf77819ca05a6b2786c76262bf7371cef97b218e96f175a3ccdda2acc058903```

Please let me know if you find that the kmackay code passes this vector and I may move back to this library.  I have to admit that burying the endianness swapping within the API, as kmackay does, has a certain appeal in removing clutter from my code.

ECDH still needs to be verified.

opi-smccoole, 2020-08-05:
@rmincher Unfortunately I'm stuck at an even more fundamental level.  Essentially right now I can't reliably compute the corresponding public key point on at least the secp256k1 and secp256r1 curves given the private key point if it is generated by OpenSSL.  I *think* it might be related to potential issues around regularizing the key.  Looking at the libecc project, they mention special handling for certain cases [here](https://github.com/ANSSI-FR/libecc/blob/master/src/curves/prj_pt.c#L878), that I don't think the micro-ecc code takes into account possibly.  Still wrapping my head around it a bit more to see if my train of thought is correct or not.

rmincher, 2020-08-05:
@opi-smccoole It seems like I was able to get that to work with this library for the r1 curve, but it's all a bit of a blur.  Using my private key from my initial post are you able to generate the public key I got (stripping off the 0x04)?

```
priv:
    5f:ef:5f:e5:62:7c:ff:60:56:87:3e:64:87:32:72:
    c3:ed:8c:08:c4:b6:01:b1:3c:36:c0:db:c2:bb:8d:
    24:fc
pub: 
    04:f1:bf:ca:c9:3c:ff:50:71:4f:7d:cc:1c:23:e7:
    03:02:d1:81:8e:88:52:20:43:01:8a:3e:68:98:78:
    9b:8a:9f:48:a2:a8:43:f4:2a:e3:3c:5a:da:d8:90:
    bc:5d:f3:87:98:64:fb:b8:ea:e7:ea:95:5e:39:84:
    3a:85:44:a3:49
```

The Intel test code for ECDH does contain the CAVP test vectors and I have confirmed the resulting Z for the first vector using pycryptodome so it looks like the Intel code works (at least once).  The vector I used was:

```
QCAVSx = 700c48f77f56584c5cc632ca65640db91b6bacce3a4df6b42ce7cc838833d287
QCAVSy = db71e509e3fd9b060ddb20ba5c51dcc5948d46fbf640dfe0441782cab85fa4ac
dIUT = 7d7dc5f71eb29ddaf80d6214632eeae03d9058af1fb6d22ed80badb62bc1a534
QIUTx = ead218590119e8876b29146ff89ca61770c4edbbf97d38ce385ed281d8a6b230
QIUTy = 28af61281fd35e2fa7002523acc85a429cb06ee6648325389f59edfce1405141
ZIUT = 46fc62106420ff012e54a434fbdd2d25ccc5852060561e68040dd7778997bd7b
```

Could someone please try this and see if it works?  I may try going back to the kmackay code purely for the hiding of the byte manipulation if it works.

opi-smccoole, 2020-08-05:
@rmincher Running that private key and extracting the public key on the r1 curve does match the value you have, so that's at least not the root of your issues it seems.  But only certain, moderately, rare private key values fail to match openSSL so I'm not too surprised that it works.

My issues with the tinycrypt code is that it only supports the secp256r1 curve as configured and I need multiple curves, plus the public key extraction is pretty much identical to what is in this version of the library so I don't think its going to help my issues.

rmincher, 2020-08-05:
@opi-smccoole We can continue this via mail if it's getting too noisy for this thread though I would expect people would be interested in any verification.

I went back to micro-ecc and it looks like that vector does work.

```
  strtobytes("44acf6b7e36c1342c2c5897204fe09504e1e2efb1a900377dbc4e7a6a133ec56", hash);
  strtobytes("1ccbe91c075fc7f4f033bfa248db8fccd3565de94bbfb12f3c59ff46c271bf83ce4014c68811f9a21a1fdb2c0e6113e06db7ca93b7404e78dc7ccd5ca89a4ca9", public1);
  strtobytes("f3ac8061b514795b8843e3d6629527ed2afd6b1f6a555a7acabb5e6f79c8c2ac8bf77819ca05a6b2786c76262bf7371cef97b218e96f175a3ccdda2acc058903", signature);
  r = uECC_verify(public1, hash, sizeof(hash), signature, curve);
  printf("**verify result = %d\n", r);

**verify result = 1
```
The nist vectors can be found at the bottom of this page:
[https://csrc.nist.gov/Projects/Cryptographic-Algorithm-Validation-Program/Component-Testing#ECDSASigGenPrim](url)

You'll have to find a source of vectors for the curves you want to support.  I don't know where sep256k1 vectors are; maybe a google search will locate some.   I'm suspecting the openssl issues are related somehow to the hash being presented to the `openssl dgst verify` in the wrong format.

motters, 2020-08-05:
All very interesting.

I'll add my progress for people following. I've come back to micro-ecc due to its flash size compared to mbedtls. I have it working with [python ecdsa](https://pypi.org/project/ecdsa/) library, albeit i have had to switch to SECP256k1.

The python library creates the keys and signs the firmware; while micro-ecc uses the public key to verify the signature. 
 
The ecdsa python lib supports the ability to load & generate openssl certs. This may help decode what is going wrong.

kmackay, 2020-10-08:
The issue is that when you run 'openssl dgst -sha256 -sign private.pem -out msg.signature.txt msg.digest.txt', it hashes whatever the input is (msg.digest.txt) in this case, and signs the result of that. If you instead run 'openssl dgst -sha256 -sign private.pem -out msg.signature.txt msg.txt', everything works as expected.

`> openssl ecparam -name prime256v1 -genkey -noout -out key.pem
> openssl ec -in key.pem -pubout -out public.pem -conv_form uncompressed
read EC key
writing EC key
> openssl ec -in key.pem -text
read EC key
Private-Key: (256 bit)
priv:
    00:ca:8f:15:46:3c:ed:57:44:69:16:37:6e:7f:32:
    ea:82:e9:52:66:47:97:cf:52:25:9e:cf:d3:ec:cb:
    9c:32
pub: 
    04:ef:54:c3:1f:04:1e:8c:d4:e7:a6:a7:2e:49:eb:
    b1:f5:8b:bb:60:01:ea:50:8d:3d:f0:3a:7e:b9:43:
    b8:63:85:75:3a:6c:48:a5:36:e7:1e:6f:dd:54:43:
    96:8d:3e:df:cb:df:15:2a:ca:be:71:8c:a4:c4:39:
    19:2f:8e:10:df
ASN1 OID: prime256v1
writing EC key
-----BEGIN EC PRIVATE KEY-----
MHYCAQEEH8qPFUY87VdEaRY3bn8y6oLpUmZHl89SJZ7P0+zLnDKgCgYIKoZIzj0D
AQehRANCAATvVMMfBB6M1Oempy5J67H1i7tgAepQjT3wOn65Q7hjhXU6bEilNuce
b91UQ5aNPt/L3xUqyr5xjKTEORkvjhDf
-----END EC PRIVATE KEY-----
> touch msg.txt | echo -n "aaa" > msg.txt
> openssl dgst -sha256 -out msg.digest.txt msg.txt
> cat msg.digest.txt
SHA256(msg.txt)= 9834876dcfb05cb167a5c24953eba58c4ac89b1adf57f28f2f9d09af107ee8f0
> openssl dgst -sha256 -sign key.pem -out msg.signature.txt msg.txt
> openssl asn1parse -inform DER -in msg.signature.txt
    0:d=0  hl=2 l=  70 cons: SEQUENCE          
    2:d=1  hl=2 l=  33 prim: INTEGER           :A0CC5254B415668A7CFE1FFCD671A72EB463656EC8060CE702459BA2BDCBD260
   37:d=1  hl=2 l=  33 prim: INTEGER           :FE086A1B54B8761DD66C1180AC94A00427451173E45F47079462F1B4FA1EBA5C
> openssl dgst -sha256 -hex -verify public.pem -signature msg.signature.txt msg.txt
Verified OK`

and then the C code using uECC to verify:

`#include "uECC.h"

#include <stdio.h>
#include <stdlib.h>
#include <string.h>

char priv[] = "00ca8f15463ced57446916376e7f32ea82e952664797cf52259ecfd3eccb9c32";
char pub[] = "ef54c31f041e8cd4e7a6a72e49ebb1f58bbb6001ea508d3df03a7eb943b86385753a6c48a536e71e6fdd5443968d3edfcbdf152acabe718ca4c439192f8e10df";
char sig[] = "A0CC5254B415668A7CFE1FFCD671A72EB463656EC8060CE702459BA2BDCBD260FE086A1B54B8761DD66C1180AC94A00427451173E45F47079462F1B4FA1EBA5C";
char hash[] = "9834876dcfb05cb167a5c24953eba58c4ac89b1adf57f28f2f9d09af107ee8f0";

void dump(const char* tag, uint8_t *vli, unsigned int size) {
  printf("%s: ", tag);
  for(unsigned i=0; i<size; ++i) {
      printf("%02X ", (unsigned)vli[i]);
  }
  printf("\n");
}

void strtobytes(const char* str, uint8_t* bytes, int count) {
  for (int c = 0; c < count; ++c) {
    if (sscanf(str, "%2hhx", &bytes[c]) != 1) {
      printf("Failed to read string to bytes");
      exit(1);
    }
    str += 2;
  }
}

int main() {
  int r;
  const struct uECC_Curve_t * curve = uECC_secp256r1();

  uint8_t private[32];
  uint8_t public[64];
  uint8_t signature[64];
  uint8_t hash_bin[32];

  uint8_t public2[64];

  strtobytes(priv, private, 32);
  strtobytes(pub, public, 64);
  strtobytes(sig, signature, 64);
  strtobytes(hash, hash_bin, 32);

  dump("private", private, sizeof(private));
  dump("public", public, sizeof(public));
  dump("signature", signature, sizeof(signature));
  dump("hash_bin", hash_bin, sizeof(hash_bin));

  uECC_compute_public_key(private, public2, curve);
  if (memcmp(public, public2, sizeof(public)) != 0) {
    printf("Public key doesn't match\n");
    return 1;
  }

  r = uECC_verify(public, hash_bin, sizeof(hash_bin), signature, curve);
  printf("verify result = %d\n", r);

  return 0;
}
`

closed 2020-10-09 by kmackay